qmk/qmk_firmware #8684: [Bug] Windows Set-up Fails with Missing Paths

mschelasin, 2020-04-05:
Near identical issue to #8601 with added complication.

Following the setup instructions on https://docs.qmk.fm/#/newbs_getting_started?id=_1-download-software

The command `qmk setup` fails with:

> INFO Found qmk_firmware at C:/Users/Martin/qmk_firmware.
INFO QMK Doctor is checking your environment.
INFO Detected Windows.
ERROR Can't find arm-none-eabi-gcc in your path.
ERROR Can't find avr-gcc in your path.
ERROR Can't find avrdude in your path.
<class 'FileNotFoundError'>
ERROR [WinError 2] The system cannot find the file specified
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 592, in __call__
    return self.__call__()
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 597, in __call__
    return self._entrypoint(self)
  File "C:/Users/Martin/qmk_firmware/lib/python/qmk/cli/doctor.py", line 230, in doctor
    bin_ok = check_binaries()
  File "C:/Users/Martin/qmk_firmware/lib/python/qmk/cli/doctor.py", line 72, in check_binaries
    if not is_executable(binary):
  File "C:/Users/Martin/qmk_firmware/lib/python/qmk/cli/doctor.py", line 159, in is_executable
    check = subprocess.run([command, version_arg], stdout=subprocess.PIPE, stderr=subprocess.PIPE, timeout=5, universal_newlines=True)
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] The system cannot find the file specified
ERROR Your build environment is not setup completely.
*** Would you like to run util/qmk_install? [y/n] y
<class 'OSError'>
ERROR [WinError 193] %1 is not a valid Win32 application
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 592, in __call__
    return self.__call__()
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 597, in __call__
    return self._entrypoint(self)
  File "C:/msys64/mingw64/lib/python3.8/site-packages/qmk_cli/subcommands/setup.py", line 45, in setup
    process = subprocess.run(['util/qmk_install.sh'])
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 1307, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
OSError: [WinError 193] %1 is not a valid Win32 application

Solution provided in comments of #8601 was to run `$ python -m pip install -U qmk`

which resulted in:

> Requirement already up-to-date: qmk in c:/msys64/mingw64/lib/python3.8/site-packages (0.0.31)
Requirement already satisfied, skipping upgrade: colorama in c:/msys64/mingw64/lib/python3.8/site-packages (from qmk) (0.4.3)
Requirement already satisfied, skipping upgrade: argcomplete in c:/msys64/mingw64/lib/python3.8/site-packages (from qmk) (1.11.1)
Requirement already satisfied, skipping upgrade: hjson in c:/msys64/mingw64/lib/python3.8/site-packages (from qmk) (3.0.1)
Requirement already satisfied, skipping upgrade: nose2 in c:/msys64/mingw64/lib/python3.8/site-packages (from qmk) (0.9.2)
Requirement already satisfied, skipping upgrade: appdirs in c:/msys64/mingw64/lib/python3.8/site-packages (from qmk) (1.4.3)
Requirement already satisfied, skipping upgrade: flake8 in c:/msys64/mingw64/lib/python3.8/site-packages (from qmk) (3.7.9)
Requirement already satisfied, skipping upgrade: yapf in c:/msys64/mingw64/lib/python3.8/site-packages (from qmk) (0.29.0)
Requirement already satisfied, skipping upgrade: six>=1.7 in c:/msys64/mingw64/lib/python3.8/site-packages (from nose2->qmk) (1.14.0)
Requirement already satisfied, skipping upgrade: coverage>=4.4.1 in c:/msys64/mingw64/lib/python3.8/site-packages (from nose2->qmk) (5.0.4)
Requirement already satisfied, skipping upgrade: mccabe<0.7.0,>=0.6.0 in c:/msys64/mingw64/lib/python3.8/site-packages (from flake8->qmk) (0.6.1)
Requirement already satisfied, skipping upgrade: pyflakes<2.2.0,>=2.1.0 in c:/msys64/mingw64/lib/python3.8/site-packages (from flake8->qmk) (2.1.1)
Requirement already satisfied, skipping upgrade: entrypoints<0.4.0,>=0.3.0 in c:/msys64/mingw64/lib/python3.8/site-packages (from flake8->qmk) (0.3)
Requirement already satisfied, skipping upgrade: pycodestyle<2.6.0,>=2.5.0 in c:/msys64/mingw64/lib/python3.8/site-packages (from flake8->qmk) (2.5.0)

then `$ cd $USERPROFILE/qmk_firmware`

then `$ git pull`

resulting in:

> Already up to date.

However, rerunning `qmk setup` still throws the same result.

## System Information

 - Keyboard:  
 - Operating system: Windows 10 Education
 - AVR GCC version: throws command not found error when trying to determine
 - ARM GCC version: throws command not found error when trying to determine
 - QMK Firmware version: 0.8.95
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

SirTimmyTimbit, 2020-04-05:
Same error here. 

Msys2 Python is 3.7. QMK is looking for 3.8. I think that's where the errors are coming from. 

I'm not sure how to upgrade python inside msys2

fauxpark, 2020-04-05:
There was a regression introduced in #8583, it should be fixed by #8678.

PR #8678 (2020-04-05): Doctor: Add avrdude/dfu-util/dfu-programmer version printing

## Description

```
255 fauxpark@marissa:qmk_firmware$ bin/qmk doctor           
Ψ QMK Doctor is checking your environment.
Ψ Detected macOS.
Ψ All dependencies are installed.
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.7.2
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 8.4.0
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

## Types of Changes

- [x] Core
- [ ] Bugfix
- [ ] New feature
- [x] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/ [... cut]
closed 2020-04-05 by Erovia