qmk/qmk_firmware #8687: [Bug] It seems audio doesn't work if the speaker is on the slave side.

pvinis, 2020-04-05:
## Describe the Bug
I'm having some strange issue. I have an iris (split). I have the right side as master. I have a speaker on the left side.

When I flash both, and then connect the usb cable to the right/master, and press a key to play sound, it doesn't play it.
If I then connect the usb cable to the left(supposedly slave), and press the same key but mirrored, it does play the sound.

Is that a limitation, or bug, or just something that was forgotten?

## System Information

 - Keyboard:  Iris
   - Revision (if applicable):  2.5
 - Operating system: macOS
 - AVR GCC version: 

 - ARM GCC version: avr-gcc (Homebrew AVR GCC 9.3.0) 9.3.0

 - QMK Firmware version: 0.8.100

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

zvecr, 2020-04-05:
As per the discussion on discord, audio wont happen when its on the slave side, due to it not processing key presses. To get both halves to process keycodes, that has its own rabbit hole of issues,
some of which are covered in #5998.

pvinis, 2020-04-13:
Indeed. I just wanted maybe some more documentation or code related to it, and the linked PR helps a bit. Thanks. I will close this issue.

closed 2020-04-13 by pvinis