qmk/qmk_firmware #8690: Hasu USB2USB - Keycode not recognised after converted to QMK

duoshock, 2020-04-05:
I use JIS keyboard and utilised '無変換' / 'JP Muhenkan' as FN key.
It works as intended with TMK firmware, but after flashing QMK, it didn't recognise the key.

I tested with other keyboard and it works, then I found out they have different keycode for the same key.

235(0xEB) - Recognised by both TMK and QMK
233(0xE9) - Only recognised by TMK

TMK firmware is generated from 'TMK Keymap Editor'

Any help please?

vomindoraan, 2020-04-11:
The 0xEB keycode corresponds to 0x8B `KC_INT5`, which is indeed the keycode for `KC_MHEN` (Muhenkan). The 0xE9 keycode, on the other hand, corresponds to 0x89 `KC_INT3`, which is actually the `KC_JYEN` (`¥|`) key that's to the left of backspace on JIS boards.

Try setting the `KC_JYEN` (`¥|`) position in your keymap as the Fn key and see if that works.

duoshock, 2020-04-17:
Thanks for your reply.
I just found out the keycode 0xE9 is Hanja 한자 key for Korean keyboard layout.
It's odd that tmk registers as JP Muhenkan.

closed 2020-04-17 by duoshock