qmk/qmk_firmware #8691: [Bug] NK65 recognized as STM Device in DFU Mode, but cannot be opened to flash

mschelasin, 2020-04-05:
Trying to Flash VIA firmware to my NK65. In toolbox I put it into reset;

> *** STM32 device connected: STMicroelectronics STM Device in DFU Mode (0483:DF11:2200)

Select the nk65_via.bin and go to flash;

> *** Attempting to flash, please don't remove device
 dfu-util.exe -a 0 -d 0483:df11 -s 0x08000000:leave -D "C:\Users\Martin\Desktop\KB Configs\NK65\nk65_via.bin"
    dfu-util 0.9
    
>  Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
    Copyright 2010-2016 Tormod Volden and Stefan Schmidt
    This program is Free Software and has ABSOLUTELY NO WARRANTY
    Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

>  Cannot open DFU device 0483:df11
    No DFU capable USB device available

## System Information

 - Keyboard:  NK65
   - Revision (if applicable):  N/A
 - Operating system: Windows 10 Education
 - AVR GCC version: 8.3.0
 - ARM GCC version: 8.3.1
 - QMK Firmware version: 0.8.95
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2020-04-05:
You have the wrong driver installed for STM32 devices. Update the Toolbox, it will prompt you and should overwrite the existing driver with the correct one.

mschelasin, 2020-04-05:
> You have the wrong driver installed for STM32 devices. Update the Toolbox, it will prompt you and should overwrite the existing driver with the correct one.

Apologies for my ignorance; new to this. How would I go about running the update? The toolbox is a clean install using the latest version as of earlier this morning.

fauxpark, 2020-04-05:
You should be running 0.0.17: https://github.com/qmk/qmk_toolbox/releases

It includes a fix to the driver installation process so that it forcibly overwrites the existing STTub30 driver with WinUSB. Did it ask you to install the drivers on first run?

mschelasin, 2020-04-05:
I did get the driver install prompt the first time around.

Did a reinstall regardless and got the prompt this time as well.

Unfortunately, still producing the same error when I go to flash though.

fauxpark, 2020-04-05:
Hm, that's annoying. In that case, run Zadig: https://zadig.akeo.ie and replace the `STM32 BOOTLOADER` device's driver with WinUSB.

mschelasin, 2020-04-05:
That got it!

Really appreciate your patience, thanks so much for the help!

closed 2020-07-06 by fauxpark