qmk/qmk_firmware #8694: [Bug] Layer gets stuck on LT(1,KC_Z)

neocsr, 2020-04-05:
In recent firmware builds, `LT(1, KC_Z)` gets stuck in `layer 1` instead of reverting back to `layer 0` when releasing the key `KC_Z`. Using the same [json configuration file](https://gist.github.com/neocsr/c7fa88163d05728f7e5af067a81605ac) that was working fine in early January, now it generates a binary with this behavior. I tried both, a remote build with QMK Configurator and a local build using Docker. The other `LT` commands work as expected. For example `LT(2,KC_A)` is working fine.

## Summary

After releasing `KC_Z`, `layer 1` remains active.

## System Information

 - Keyboard: `hotdox`
 - Operating system: `macOS Catalina`
 - AVR GCC version: `avr-gcc (GCC) 5.4.0`
 - ARM GCC version: `arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors 6-2017-q2-update) 6.3.1 20170620 (release) [ARM/embedded-6-branch revision 249437]`
 - QMK Firmware version: `0.8.101`

## Additional Context

- JSON file used in QMK Configurator: [hotdox.json](https://gist.github.com/neocsr/c7fa88163d05728f7e5af067a81605ac)
- Keymap file: [keymap.c](https://gist.github.com/neocsr/ac84033d81e01fbc49f1ed337e628981)
 
Flash output:

```
*** DFU device connected
*** Attempting to flash, please don't remove device
>>> dfu-programmer atmega32u4 erase --force
    Erasing flash...  Success
    Checking memory from 0x0 to 0x6FFF...  Empty.
>>> dfu-programmer atmega32u4 flash /Users/user/qmk_firmware/hotdox.hex
    Checking memory from 0x0 to 0x597F...  Empty.
    0%                            100%  Programming 0x5980 bytes...
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    0%                            100%  Reading 0x7000 bytes...
    [>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>>]  Success
    Validating...  Success
    0x5980 bytes written into 0x7000 bytes memory (79.91%).
>>> dfu-programmer atmega32u4 reset
    
*** DFU device disconnected
*** Z.P.Z.O. - HotDox connected -- FEED:E11D

```

zvecr, 2020-04-05:
Same issue as #8600, see that issue for `KC_TRANS` workarounds.

closed 2020-04-05 by zvecr

neocsr, 2020-04-06:
Tested with latest commits `927b4fe4c` and `68edb1910` from https://github.com/qmk/qmk_firmware/pull/8693/commits and the issue is fixed.