qmk/qmk_firmware #8733: [Bug] Infinity60/Whitefox slow startup time in Linux

andrea-pretto-rawfish, 2020-04-08:
I just updated qmk firmare on both my Infinity60 and Whitefox and I noticed that when I plug in the keyboard it takes 10/20 seconds to the keyboard to send key events to the OS.

Before updating I had a very old qmk version and I did not have this problem.
I have another Planck keyboard and it works fine, so maybe it's something related to chibios??

I tried on macos and it's much faster, it takes 2/3 seconds to become reponsive.

## System Information

 - Keyboard:  
   - Infinity 60
   - Whitefox  
 - Operating system:
Fedora with Linux kernel: 4.9.13-101.fc24.x86_64

 - AVR GCC version: 
avr-gcc (Fedora 6.2.0-1.fc24) 6.2.0
 - ARM GCC version: 
arm-none-eabi-gcc (Fedora 6.2.0-2.fc24) 6.2.0
 - QMK Firmware version:
0.8.103

## Additional Context

This is the output from dmesg:
```
[  715.055001] usb 1-2: new full-speed USB device number 12 using xhci_hcd
[  715.224977] usb 1-2: New USB device found, idVendor=1c11, idProduct=b04d
[  715.224980] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  715.224982] usb 1-2: Product: Infinity_60%_LED/QMK
[  715.224984] usb 1-2: Manufacturer: Input Club
[  715.224986] usb 1-2: SerialNumber: 0
[  715.227899] input: Input Club Infinity_60%_LED/QMK as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.0/0003:1C11:B04D.0017/input/input33
[  715.281050] hid-generic 0003:1C11:B04D.0017: input,hidraw1: USB HID v1.11 Keyboard [Input Club Infinity_60%_LED/QMK] on usb-0000:00:14.0-2/input0
[  725.728390] hid-generic 0003:1C11:B04D.0018: usb_submit_urb(ctrl) failed: -1
[  725.728422] hid-generic 0003:1C11:B04D.0018: timeout initializing reports
[  725.728789] input: Input Club Infinity_60%_LED/QMK as /devices/pci0000:00/0000:00:14.0/usb1/1-2/1-2:1.1/0003:1C11:B04D.0018/input/input34
[  725.781476] hid-generic 0003:1C11:B04D.0018: input,hidraw2: USB HID v1.11 Mouse [Input Club Infinity_60%_LED/QMK] on usb-0000:00:14.0-2/input1
[  735.968269] hid-generic 0003:1C11:B04D.0019: timeout initializing reports
[  735.969182] hid-generic 0003:1C11:B04D.0019: hiddev0,hidraw3: USB HID v1.11 Device [Input Club Infinity_60%_LED/QMK] on usb-0000:00:14.0-2/input2
```

It seems that the keyboard become responsive after the last line in dmesg.

ToyKeeper, 2021-04-23:
Same issue here.  When connected, it takes about 12 seconds to reach a usable state.  It takes ~2 seconds to get noticed, then another 10 seconds to start working.  Here's what shows up in syslog.  It was physically plugged in at 03:28:39.

```
Apr 23 03:28:30 banana kernel: [18876570.368003] usb 1-6: USB disconnect, device number 92
Apr 23 03:28:41 banana kernel: [18876581.156843] usb 1-6: new full-speed USB device number 93 using xhci_hcd
Apr 23 03:28:41 banana kernel: [18876581.298190] usb 1-6: New USB device found, idVendor=1c11, idProduct=b04d
Apr 23 03:28:41 banana kernel: [18876581.298199] usb 1-6: New USB device strings: Mfr=1, Product=2, SerialNumber=0
Apr 23 03:28:41 banana kernel: [18876581.298204] usb 1-6: Product: WhiteFox (QMK)
Apr 23 03:28:41 banana kernel: [18876581.298208] usb 1-6: Manufacturer: Input:Club
Apr 23 03:28:41 banana kernel: [18876581.300617] input: Input:Club WhiteFox (QMK) as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.0/0003:1C11:B04D.0059/input/input102
Apr 23 03:28:41 banana kernel: [18876581.357097] hid-generic 0003:1C11:B04D.0059: input,hidraw0: USB HID v1.11 Keyboard [Input:Club WhiteFox (QMK)] on usb-0000:00:14.0-6/input0
Apr 23 03:28:51 banana kernel: [18876591.468608] hid-generic 0003:1C11:B04D.005A: usb_submit_urb(ctrl) failed: -1
Apr 23 03:28:51 banana kernel: [18876591.468627] hid-generic 0003:1C11:B04D.005A: timeout initializing reports
Apr 23 03:28:51 banana kernel: [18876591.468761] input: Input:Club WhiteFox (QMK) as /devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/0003:1C11:B04D.005A/input/input103
Apr 23 03:28:51 banana kernel: [18876591.529152] hid-generic 0003:1C11:B04D.005A: input,hidraw1: USB HID v1.11 Mouse [Input:Club WhiteFox (QMK)] on usb-0000:00:14.0-6/input1
Apr 23 03:28:51 banana mtp-probe: checking bus 1, device 93: "/sys/devices/pci0000:00/0000:00:14.0/usb1/1-6"
Apr 23 03:28:51 banana mtp-probe: bus: 1, device: 93 was not an MTP device
Apr 23 03:28:51 banana systemd-logind[558]: Watching system buttons on /dev/input/event0 (Input:Club WhiteFox (QMK))
Apr 23 03:28:51 banana systemd-logind[558]: Watching system buttons on /dev/input/event1 (Input:Club WhiteFox (QMK))
```

In particular, this part look suspicious:
```
hid-generic 0003:1C11:B04D.005A: usb_submit_urb(ctrl) failed: -1
hid-generic 0003:1C11:B04D.005A: timeout initializing reports
```

This might not help much, but at least it's a +1 confirming the issue still exists a year later.

jackhumbert, 2021-10-14:
I was having the same issue on my Linux machine - does #14814 address it for you?