qmk/qmk_firmware #8763: RFC Proposal to rename debounce algorithms

purdeaandrei, 2020-04-11:
Please see current naming scheme description here:
https://github.com/qmk/qmk_firmware/blob/master/quantum/debounce/readme.md

The problem with it is the Sym/Asym and Eager/Non-Eager concepts are orthogonal, and the concepts are mixed in the current naming scheme.

So I would like to separate out section "2.Eager vs symmetric vs asymmetric" into two different section: 2. Eager/Non-Eager, and 3. Symmetric/Assymetric.

I am not sure what is the best name to call the opposite of Eager, i'm thinking of the following words:
1) waitfirst
2) noneager
3) lazy

Lazy looks good, but waitfirst is more descriptive. I will use waitfirst in my below examples, but I'm
open for any suggestions.

Also I would like to include Timestamp/Cycles in the name of the algorithm. I agree that timestamp-based algorithm are superior for normal physical switches, however in some cases where is noise present in the absence of real user input, and if the keyboard scan rate is not very high, the existence of cycles-based algorithms may be justified. And to minimize confusion when I will implement a cycle-based algorithm next, I propose to change the naming scheme to avoid such confusion. ts_ would be timestamp, while cy_ would be cycles-based.

So I propose the following renames:

* sym_g    -> ts_sym_waitfirst_g
* eager_pk -> ts_sym_eager_pk
* sym_pk   -> ts_sym_waitfirst_pk
* eager_pr -> ts_sym_eager_pr

Also I would keep the old names as well, to be interpreted as the new name using Makefile code, but it should also add a $(warning ) that the old names will be deprecated in a future breaking change.

And in the readme.md, I would add the following examples for other names of future algorhythms:

* ts_sym_eager_g
* ts_asym_eagerdown_waitfirstup_pk
* cy_sym_pk ( I intend to implement this one eventually, as I would like to use it in one of my keyboards )

purdeaandrei, 2020-04-11:
Note: I intend to do the Pull Request for this but before I do it I'd like to see agreement on the naming scheme.

vomindoraan, 2020-04-11:
I'm only familiar with the debouncing implementations at a high level, so take this with a grain of salt. But to me it seems that `eager`/`lazy` would be a misnomer here, since that usually refers to something being done _before its needed_ (eager) vs _exactly when its needed_ (lazy), and those two terms are usually paired together. I don't think that translates well to this use case, as key reporting isn't something that is needed at any exact point, but rather it's a part of the algorithm to decide when it happens, whether before or after the wait. So I think something like `waitafter`/`waitbefore` or `waitlast`/`waitfirst` would be more accurate.

However, since `eager` in the context of debouncing seems to already be established terminology in QMK that people are familiar with, it might be a good idea to keep it. In that case, I propose `eager`/`defer` as the naming convention, since in the “wait first” case the key reporting is _deferred_ to occur after the debounce interval.

purdeaandrei, 2020-04-11:
@vomindoraan "defer" - I like it, I didn't think of that, I'll edit my proposal

zvecr, 2020-04-11:
As a implementation note: Given the way debounce is hooked up, this will either need to go via the [breaking changes](https://docs.qmk.fm/#/breaking_changes) process (or backward compatibility introduced at the algorithm selection processing).

purdeaandrei, 2020-04-11:
@zvecr 
> or backward compatibility introduced at the algorithm selection processing

Yes, backwards compatibility is my intent. I would add a $(warning) message though, if the old name is used.

zvecr, 2020-04-11:
I'm personally not keen on formally deprecating stuff outside the breaking changes window (which I would classify adding the build warning as).

purdeaandrei, 2020-04-11:
> I'm personally not keen on formally deprecating stuff outside the breaking changes window (which I would classify adding the build warning as).

Alright, noted. In that case I won't add $(warning)

alex-ong, 2020-04-23:
@zvecr wrt deprecation due to renames:

Since its a simple rename, wouldn't going through people's rules.mk and updating it in the same PR be sufficient? Or is that a general no-no in QMK?

IIRC 99% of them use whatever default we provide them (ts_sym_defer_g) so there wouldn't be much change there.

Or we could backward compatible for a short period before `breaking changing` and removing + auto-upgrading peoples rules.mk.

I think that having two names for the same debounce algorithm would be confusing, and i'd rather we just enforce the new, better one.

> * ts_asym_eagerdown_deferup_pk
I think we could simplify it to 
`ts_asym_eager_defer_pk` (keeps all asym algorithms together when sorting by name)
or
`ts_eager_defer_pk` (shorter)

And just rely on it always being in KeyDownReportMethod/KeyUpReportMethod order.

purdeaandrei, 2020-05-02:
Edited top description to remove the -down/-up suffix from the assymetric algorithm names, according to @alex-ong 's suggestion.

purdeaandrei, 2020-05-02:
I've implemented the following pull-request, please review: https://github.com/qmk/qmk_firmware/pull/8998

purdeaandrei, 2020-05-02:
I'm not sure if this will be final.
I'm thinking of two things now:
1) I think cycles variants would be best implemented by replacing timer_read() with something that returns the current cycle number instead, to avoid code duplication. In this case, I think the cycles/timestamp should not be part of the name, but instead should be another configuration option.
Otherwise there will be lots of slightly similar looking files, with lots of code duplication.
So i'm thinking of not adding the ts_ and cy_ prefixes, and adding DEBOUNCE_UNIT makefile variable, which will be traduced into a -D macro:
   * ```DEBOUNCE_UNIT=MS``` (default)
   * ```DEBOUNCE_UNIT=CYCLES```
2) I'm also thinking that we should give the option of selecting different debounce lengths for key-down/key-up, but using the same algorithms (this would make sense for per-key algorithms, and for eager algorithms). I'm wondering should that be considered symmetric or assymetric? I'm thinking it should still be called a symmetric algorithm, because this specific type of assimetry will be implemented right in the .c file, rather then making a copy. It makes no sense to have ```asym_defer_defer_pk.c``` for example, just to customize the keydown/keyup debounce timeouts.
What I would to is to have something like:
````
#ifndef DEBOUNCE_KEYDOWN
#define DEBOUNCE_KEYDOWN DEBOUNCE
#endif
#ifndef DEBOUNCE_KEYUP
#define DEBOUNCE_KEYUP DEBOUNCE
#endif

... And later in the actual debounce code, for example in eager_pk i would have: ...

#if DEBOUNCE_KEYDOWN == DEBOUNCE_KEYUP
if (TIMER_DIFF(current_time, *debounce_pointer, MAX_DEBOUNCE) >= DEBOUNCE_KEYDOWN) {
#else
if (
((cooked[row] & (ROW_SHIFTER << col)) && (TIMER_DIFF(current_time, *debounce_pointer, MAX_DEBOUNCE) >= DEBOUNCE_KEYDOWN)) ||
((!(cooked[row] & (ROW_SHIFTER << col))) && (TIMER_DIFF(current_time, *debounce_pointer, MAX_DEBOUNCE) >= DEBOUNCE_KEYUP)) ) {
#endif
```

Any opinions?

alex-ong, 2020-05-03:
1. Yes i think we could make it a macro instead, with `timer_read()` or `cycle_count` being wrapped in a new `get_timer()`. Since debounce is called once per frame we can add `cycle_count += 1` somewhere, maybe we need an abstract debounce method for everything to inherit off, where the base method adds 1 to cycle count, and also wraps the `timer_read()` / `cycle_count` behavior.

2. I initially had speed concerns but that sample code clears my thoughts. Monolithic debounce algorithm with multiple settings is better than modular algorithms where 90% of the code is copy pasted.

Someone was also doing a PR for a hybrid cycles/ms pattern... i'll try and find and link it.

drashna, 2020-05-09:
> Since its a simple rename, wouldn't going through people's rules.mk and updating it in the same PR be sufficient? Or is that a general no-no in QMK?

It's fine. But renaming it like that would likely means that it needs to got through the breaking change process.  Which means more time for it to go through, basically.  

The best bet is to add mapping now, and then later, do a cleanup pass, removing the old naming. This way, the change gets in faster, and the renaming goes through breaking changes. 

It's a bit more work, but ...

purdeaandrei-movi, 2020-05-09:
Is changing the name used in current keyboards in the repository a breaking change?
I would have thought that dropping support for the old names would be a breaking change which, is fine. But renaming the existing keyboards to use the new names should not be breaking. There were 8 rules.mk files that I touched in my PR, do you want me to not do those changes to make my change not-breaking?

purdeaandrei-movi, 2020-05-09:
@drashna

zvecr, 2020-05-10:
<https://github.com/qmk/qmk_firmware/pull/8998#issuecomment-626257819>

If you want to avoid breaking changes, you will need to revert keymap changes and provide backwards compatibility.

purdeaandrei, 2020-06-28:
Added new PR here: https://github.com/qmk/qmk_firmware/pull/9564

Removed the ts_ and cy_ prefixes as they will be implemented with a macro, and not separate files,
and also removed the touching of individual keyboards, to avoid breaking changes..

I will implement work on DEBOUNCE_UNIT and DEBOUNCE_KEYDOWN/DEBOUNCE_KEYUP after this is is merged,
as separate reviewable changesets.

closed 2020-10-26 by stale[bot]