qmk/qmk_firmware #8803: [Bug] Oneshit layer doesn't return with unicode map

AnonJohn, 2020-04-14:
## Describe the Bug
I use unicode map in a layer, which is accessed as a oneshot layer [osl function]. XP(i,j) works as expected, but OSL does not; it doesn't return to the original layer. Standard keys on the OSL work as expected, with a one shot and return to the original layer.

This is hopefully readily worked around with other unicode features and another layer, but that will consume extra bytes!

## System Information

 - Keyboard:  Ergodox EZ
   - Revision (if applicable):  
 - Operating system: Linux (Mint)
 - AVR GCC version: 5.4.0

 - ARM GCC version: (15:6.3.1+svn253039-1build1) 6.3.1 20170620

 - QMK Firmware version: 0.8.17

 - Any keyboard related software installed? No
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

spidey3, 2020-04-18:
@AnonJohn - I am doing some work on One Shot Layers, and I use the UNICODEMAP /  XP() functionality, but I cannot reproduce this issue, even with the code before my changes.  Can you share your code?

AnonJohn, 2020-04-18:
> @AnonJohn - I am doing some work on One Shot Layers, and I use the UNICODEMAP / XP() functionality, but I cannot reproduce this issue, even with the code before my changes. Can you share your code?

Sure, happy to, though it's really an embarrassing mess in development. Not as embarrassing as my title typo (sorry, thanks!), but still. I think I attached as a zip. Let me know if you want it a different way. Note that this does not exhibit the issue, because of a work-around at the end of process_user_record. If you comment that out the issue resurfaces. 

Work around:

     if ((keycode & QK_UNICODEMAP) == QK_UNICODEMAP && record->event.pressed) {
	layer_off(6); //should be oneshot
	layer_off(7); //should be oneshot
	process_unicode_common(keycode,record);
	return false;
      }

[ergoqmk.zip](https://github.com/qmk/qmk_firmware/files/4496719/ergoqmk.zip)

Cheers + thanks!

spidey3, 2020-04-18:
@AnonJohn - that's a pretty complicated keymap! I tried to mimic what you are doing, but I still cannot reproduce.  My suggestion would be to create a stripped down version, that just has two layers - 0 and 1, with an OSL(1) in the 0 layer to trigger the 1 layer, which has your XP()'s. Remove everything else, and let's see if you still see the issue.  Also, I wonder, how long ago did you last do a pull from upstream?

AnonJohn, 2020-04-20:
@spidey3 Sage advice, and I will take it, though it will probably be a few weeks. (If that hits your plans, let me know and I'll try to get to it sooner). My last pull was on March 6, so not super current. Think it would make a difference?

spidey3, 2020-04-20:
Hmm - a new pull might help...

AnonJohn, 2020-04-27:
Happy to report that a new pull fixed it. I'd mark it as closed. Thanks for the help. Sorry I can't point to a root cause, but glad it works as expected now!

closed 2020-04-27 by AnonJohn