qmk/qmk_firmware #8809: arm-none-eabi-gcc error on Arch Linux

galmio, 2020-04-15:
## Describe the Bug

I've been trying to compile and flash both a dz60rgb ansi v1 and a dz65rgb v1 in Arch Linux using the [qmk AUR package](https://aur.archlinux.org/packages/qmk/)  without success due to this error when trying to do so:
```bash
arch% qmk compile
Ψ Compiling keymap with make dztech/dz65rgb/v1:adi

QMK Firmware 0.8.120
Making dztech/dz65rgb/v1 with keymap adi

make[1]: Entering directory '/home/{my_user}/qmk_firmware'
arm-none-eabi-gcc (Arch Repository) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Linking: .build/dztech_dz65rgb_v1_adi.elf                                                           [ERRORS]
 | 
 | /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-abort.o): in function `abort':
 | abort.c:(.text.abort+0xa): undefined reference to `_exit'
 | /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-
signalr.o): in function `_kill_r':
 | signalr.c:(.text._kill_r+0xe): undefined reference to `_kill'
 | /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-signalr.o): in function `_getpid_r':
 | signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
 | collect2: error: ld returned 1 exit status
 | 
make[1]: *** [tmk_core/rules.mk:306: .build/dztech_dz65rgb_v1_adi.elf] Error 1
make[1]: Leaving directory '/home/{my_user}/qmk_firmware'
Make finished with errors
make: *** [Makefile:579: dztech/dz65rgb/v1:adi] Error 1
```
I tried downgrading `arm-none-eabi-gcc` back to `8.3` as [recommended on this issue](https://github.com/qmk/qmk_firmware/issues/7964#issuecomment-577210889) to no effect. I also tried removing all the arm tools dependencies and then running `./util/qmk_install.sh` but it did not help either.

## System Information

 - Keyboard:  dz65rgb v1, dz60rgb v1
   - Revision (if applicable):  
 - Operating system: Arch Linux 5.6.3
 - AVR GCC version: 8.3.0

 - ARM GCC version: 8.3.0 and 9.3.0

 - QMK Firmware version: 0.8.120

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

datNoko, 2020-04-15:
Can you post the output you get with arm-none-eabi-gcc 8.3 or is it the same?

galmio, 2020-04-15:
```bash
Linking: .build/dztech_dz65rgb_v1_adi.elf                                                           [ERRORS]
 | 
 | /usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-abort.o): in function `abort':
 | abort.c:(.text.abort+0xa): undefined reference to `_exit'
 | /usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-signalr.o): in function `_kill_r':
 | signalr.c:(.text._kill_r+0xe): undefined reference to `_kill'
 | /usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/8.3.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-signalr.o): in function `_getpid_r':
 | signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
 | collect2: error: ld returned 1 exit status
 | 
make[1]: *** [tmk_core/rules.mk:306: .build/dztech_dz65rgb_v1_adi.elf] Error 1
make[1]: Leaving directory '/home/galmio/qmk_firmware'
Make finished with errors
make: *** [Makefile:579: dztech/dz65rgb/v1:adi] Error 1
```

fauxpark, 2020-04-15:
Make sure your repo is up to date, and run `make git-submodule`.

galmio, 2020-04-15:
I tried that with both keyboards to no effect. I've tested out other dztech keyboards and it seems to only be happening with the v1 variants, so `dztech/dz60rgb/v1`, `dztech/dz60rgb_ansi/v1`, `dztech/dz60rgb_wkl/v1` and `dztech/dz65rgb/v1` have the same issue on compilation while their v2 counterparts don't.

fauxpark, 2020-04-15:
Those boards are ARM, whereas the v2 ones are AVR, so that would be why.

Interestingly, I tried compiling the dz60rgb/v1 myself, and got this:
```
Linking: .build/dztech_dz60rgb_v1_default.elf                                                       [ERRORS]
 | 
 | /usr/local/Cellar/arm-gcc-bin@8/8-2019-q3-update/bin/../lib/gcc/arm-none-eabi/8.3.1/../../../../arm-none-eabi/bin/ld: cannot open linker script file ./lib/chibios/os/common/startup/ARMCMx/compilers/GCC/ld/.ld: No such file or directory
 | collect2: error: ld returned 1 exit status
 | 
make[1]: *** [.build/dztech_dz60rgb_v1_default.elf] Error 1
make: *** [dztech/dz60rgb/v1:default] Error 1
Make finished with errors
```

fauxpark, 2020-04-15:
Can you post the output of `git submodule status`?
Mine is:
```
 313416b8fda90d9973a749a0a35970956852c286 lib/chibios (breaking_2020_q1)
 e3a3a24047717af33b098266e11c2e4e5102206a lib/chibios-contrib (breaking_2020_q1)
 ec44c6c1675c25b9827aacd08c02433cccde7780 lib/googletest (release-1.8.0)
 ce10f7642b0459e409839b23cc91498945119b4d lib/lufa (LUFA-170418-97-gce10f76)
 40b48f470addad6a4fb1177de1a69a181158739b lib/ugfx (v2.7-203-g40b48f47)
```

galmio, 2020-04-15:
```bash
313416b8fda90d9973a749a0a35970956852c286 lib/chibios (breaking_2020_q1)
 e3a3a24047717af33b098266e11c2e4e5102206a lib/chibios-contrib (breaking_2020_q1)
 ec44c6c1675c25b9827aacd08c02433cccde7780 lib/googletest (release-1.8.0)
 ce10f7642b0459e409839b23cc91498945119b4d lib/lufa (LUFA-170418-107-gce10f7642)
 40b48f470addad6a4fb1177de1a69a181158739b lib/ugfx (v2.7-203-g40b48f47)
```

fauxpark, 2020-04-15:
Could you try a `make distclean` first?

galmio, 2020-04-15:
I tried, but the output is the same. Also used `make dztech/dz60rgb_ansi/v1:default:dfu-util but I receive the same error. Do you think it could have anything to do with the [AUR package](https://aur.archlinux.org/packages/qmk/)

datNoko, 2020-04-15:
> I tried, but the output is the same. Also used `make dztech/dz60rgb_ansi/v1:default:dfu-util` but I receive the same error. Do you think it could have anything to do with the [AUR package](https://aur.archlinux.org/packages/qmk/)?
Just for testing i downloaded the AUR Package and tried it with this but it worked for me.

galmio, 2020-04-15:
This is my output of `qmk doctor`:
```bash
arch% qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ Detected Linux.
⚠ Missing udev rules for 'dfu' boards. You'll need to use `sudo` in order to flash them.
⚠ Missing udev rules for 'tmk' boards. You'll need to use `sudo` in order to flash them.
⚠ Missing udev rules for 'input_club' boards. You'll need to use `sudo` in order to flash them.
⚠ Missing udev rules for 'stm32' boards. You'll need to use `sudo` in order to flash them.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.3.0
Ψ Found avr-gcc version 8.3.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```
I also saw [this instructions](https://github.com/qmk/qmk_firmware/blob/master/docs/faq_build.md) but I can't seem to compile the keyboards with `sudo` either. I really don't know how to reproduce the issue.

Mesysz, 2020-04-15:
I have the same problem right now (using dz60rgb_ansi):
`qmk compile -kb dztech/dz60rgb_ansi/v1 -km default`
returns
```
Ψ Compiling keymap with make dztech/dz60rgb_ansi/v1:default

QMK Firmware 0.8.120
Making dztech/dz60rgb_ansi/v1 with keymap default

make[1]: Entering directory '/home/{myuser}/qmk_firmware'
arm-none-eabi-gcc (Arch Repository) 9.3.0
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Linking: .build/dztech_dz60rgb_ansi_v1_default.elf                                                  [ERRORS]
 |
 | /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-abort.o): in function `abort':
 | abort.c:(.text.abort+0xa): undefined reference to `_exit'
 | /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-signalr.o): in function `_kill_r':
 | signalr.c:(.text._kill_r+0xe): undefined reference to `_kill'
 | /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.3.0/../../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-signalr.o): in function `_getpid_r':
 | signalr.c:(.text._getpid_r+0x0): undefined reference to `_getpid'
 | collect2: error: ld returned 1 exit status
 |
make[1]: *** [tmk_core/rules.mk:306: .build/dztech_dz60rgb_ansi_v1_default.elf] Error 1
make[1]: Leaving directory '/home/{myuser}/qmk_firmware'
Make finished with errors
make: *** [Makefile:579: dztech/dz60rgb_ansi/v1:default] Error 1
```
although when running it with v2 I get everything okay. The problem is.. I need V1.
`qmk doctor`:
```
Ψ QMK Doctor is checking your environment.
Ψ Detected Linux.
⚠ Missing udev rules for 'dfu' boards. You'll need to use `sudo` in order to flash them.
⚠ Missing udev rules for 'tmk' boards. You'll need to use `sudo` in order to flash them.
⚠ Missing udev rules for 'input_club' boards. You'll need to use `sudo` in order to flash them.
⚠ Missing udev rules for 'stm32' boards. You'll need to use `sudo` in order to flash them.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.3.0
☒ We do not recommend avr-gcc newer than 8. Downgrading to 8.x is recommended.
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ Problems detected, please fix these problems before proceeding.
```
I will downgrade avr-gcc later, but as I saw it does not work now so I wrote that before downgrading.

Erovia, 2020-04-15:
Something is off with Arch's arm-none-eabi-gcc bundle because I cannot recreate it on Fedora with either 9.2 or 8.3.  

In the meantime, could you please try to build you firmware with `make dztech/dz60rgb_ansi/v1:adi EXTRAFLAGS+=--specs=nosys.specs` and see if anything breaks?

Mesysz, 2020-04-15:
I have this:
```
QMK Firmware 0.8.120
Making dztech/dz60rgb_ansi/v1 with keymap default and target adi                               [ERRORS]
make[1]: Entering directory '/home/{user}/qmk_firmware'
make[1]: *** No rule to make target 'adi'.  Stop.
make[1]: Leaving directory '/home/{user}/qmk_firmware'
Making dztech/dz60rgb_ansi/v1 with keymap via and target adi                                   [ERRORS]
make[1]: Entering directory '/home/{user}/qmk_firmware'
Enabling LTO on ChibiOS-targeting boards is known to have a high likelihood of failure.
If unsure, set LINK_TIME_OPTIMIZATION_ENABLE = no.
make[1]: *** No rule to make target 'adi'.  Stop.
make[1]: Leaving directory '/home/{user}/qmk_firmware'
Make finished with errors
make: *** [Makefile:579: dztech/dz60rgb_ansi/v1:adi] Error 1
```
both on 7.x and 9.x

Erovia, 2020-04-15:
@Mesysz becaues you don't have galmio's keymap.
Try `make dztech/dz60rgb_ansi/v1:default EXTRAFLAGS+=--specs=nosys.specs`.

galmio, 2020-04-15:
It works on my side with `EXTRAFLAGS`. Should I close the issue?

Mesysz, 2020-04-16:
`make dztech/dz60rgb_ansi/v1:default EXTRAFLAGS+=--specs=nosys.specs` works.
I was tired and didn't see the "no rule to make target adi", also I didn't think about changing it to "default".
It works on my side with `EXTRAFLAGS`, too. Thank you for help @Erovia and sorry for interrupting, @galmio. :)

Erovia, 2020-04-16:
> It works on my side with `EXTRAFLAGS`. Should I close the issue?

No, as the underlying issue still needs to be fixed in either Arch's arm-none-eabi-gcc or our build scripts.

Mesysz, 2020-04-16:
Is there any option to pass `EXTRAFLAGS` to `qmk flash`? Because I can `make` it correctly, but flashing tries to build it without flags and gives the same error

galmio, 2020-04-16:
You can also do `make dztech/dz60rgb_ansi/v1:default:flash` to flash it according to [these instructions](https://beta.docs.qmk.fm/developing-qmk/qmk-reference/getting_started_make_guide)

Erovia, 2020-04-16:
Yes, `make dztech/dz60rgb_ansi/v1:default:flash EXTRAFLAGS+=--specs=nosys.specs` should also work, but the `rules.mk` file is probably more convenient.

tzarc, 2020-04-27:
Can confirm that this is now occurring on Ubuntu 20.04 as well.
`EXTRAFLAGS+=--specs=nosys.specs` does get things compiling. Output untested.

closed 2020-05-08 by zvecr

sundbp, 2020-10-01:
Leaving a breadcrumb for anyone else on Arch/Manjaro. I got it working downgrading to 8.3 like this:
[image]

PR #9044 (2020-05-08): Revert "Switch to nano specs on ChibiOS builds"
Reverts qmk/qmk_firmware#8270

Fixes #8809

`syscalls_cpp.cpp` contains some of the required stub functions, however on this arch environment it produces:
* ```multiple definition of `__dso_handle'```
* +2k firmware size over nosys

For the most part, ARM users are not strapped for space. So I put the case that it is better to have working builds across the board in the short term, while a potential longer term solution is worked on.