qmk/qmk_firmware #8878: [Bug] KC_NO ignored for keys with unicode input

karlc1, 2020-04-22:
## Describe the Bug

I have a layer where I use unicode input for some keys, like this: `XP(aa, AA)`. I also have another layer where most keys are disabled using `KC_NO`, which is defined for the same keys that has unicode input at the other layer. If the active layer is the one with unicode input, and I then switch to the layer with KC_NO, the unicode characters still ouputs as if I was still on the previous layer when I press them. 

## System Information

 - Keyboard:  Iris from keeb.io
   - Revision (if applicable):  4
 - Operating system:
 - AVR GCC version: 5.4.0

 - ARM GCC version: arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620

 - QMK Firmware version: 0.6.439

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
I compile and flash using docker if that is relevant.

ridingqwerty, 2020-04-23:
Do you have a link to your keymap @karlc1 ?

karlc1, 2020-04-24:
Sure, here it is: https://github.com/karlc1/qmk_firmware/blob/master/keyboards/keebio/iris/keymaps/karlkeyboard/keymap.c

ridingqwerty, 2020-04-24:
I think what’s going on here is that you’re using `TG` to turn layers on.  This is okay, but what I think is happening is you turn on layer 4, and then without toggling layer 4 off you turn on layer 3 so that both are active and layer 4 keycodes take precedence.

karlc1, 2020-04-27:
I had no idea that layers could be active simultaneously. I will try to verify your theory tomorrow, however, if it does it still holds true that the XP keycodes are handled differently than KC given that only those still produce input.

closed 2022-08-28 by zvecr