qmk/qmk_firmware #8946: [Feature Request] Disable autoshift on a layer

Yamakaky, 2020-04-27:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

On my Ergodox EZ, I have a letter layer, a special keys layer (alt-grey in french azerty), a media layer (pause, volume control...) and a gaming layer with only simple keys (no tap-hold or double function keys). I'd like auto-shift to be automatically enabled on the first two layers but not the later two. I don't think this possible currently, you have to toggle it manually. Would it be possible to make it configurable, maybe with a blacklist/whitelist of layers?

Yamakaky, 2020-04-27:
Tagging @ezuk after our mail discussion.

IsaacElenbaas, 2020-05-08:
I did this by modifying my layer toggles, but it gets complicated fast if there are multiple ways to get out of any layer. Would be *very* convenient and great for code readability.

IsaacElenbaas, 2020-06-03:
Pretty easy implementation now that I think about it - just check layer [here](https://github.com/qmk/qmk_firmware/blob/master/quantum/process_keycode/process_auto_shift.c#L77) and return true, no need to actually enable/disable AutoShift on layer status changes. Not sure what a clean way of specifying the non-autoshifted layers would be, though.

Nadrieril, 2020-06-07:
You might be able to enable/disable autoshift from your keymap file like this:
```c
uint32_t layer_state_set_user(uint32_t state) {
    if (autoshift_enabled && (state & (1<<MEDIA_LAYER) || state & (1<<GAMING_LAYER))) {
        autoshift_disable();
    } else {
        autoshift_enable();
    }
    ...
    return state;
};
```
I haven't tested that but I use something similar to have a custom swap_hands layer which changes some keys in addition to enabling the swap_hands feature.

michaelduminy, 2020-10-03:
I would also like this feature

closed 2020-11-03 by stale[bot]

byamagata, 2021-11-16:
Are there any other requests like this that I can try and find for a solution?  This would be great and I would love to see this happen!