qmk/qmk_firmware #8973: 'gbk' codec can't decode byte

xiaoquanaa, 2020-04-29:
$ qmk setup
ERROR qmk_firmware not found!
*** Would you like to clone qmk/qmk_firmware? [y/n] y
正克隆到 '/c/Users/zlx/qmk_firmware'...
error: RPC 失败。curl 56 OpenSSL SSL_read: Connection closed abruptly, errno 0 (Fatal because this is a curl debug build)
<class 'UnicodeDecodeError'>
ERROR 'gbk' codec can't decode byte 0xab in position 11: illegal multibyte sequence
Traceback (most recent call last):
  File "D:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 608, in __call__
    return self.__call__()
  File "D:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 613, in __call__
    return self._entrypoint(self)
  File "D:/msys64/mingw64/lib/python3.8/site-packages/qmk_cli/subcommands/setup.py", line 39, in setup
    clone(git_url, cli.args.destination, cli.config.setup.branch)
  File "D:/msys64/mingw64/lib/python3.8/site-packages/qmk_cli/git.py", line 24, in clone
    for line in p.stdout:
UnicodeDecodeError: 'gbk' codec can't decode byte 0xab in position 11: illegal multibyte sequence

zvecr, 2020-04-29:
@skullydazed @Erovia Any ideas on this one?

skullydazed, 2020-04-29:
I've just released a new version of the CLI, 0.0.34, which I believe fixes this. Can you upgrade to that and try again?

Gladosxiao, 2020-05-27:
hi guys,

I faced the same issue today,

I download qmk_firmware for github (2020-5-27 09:23:51), unzip and rename to "qmk_firmware" , then path the folder into "C:/Users/GlaDo/". 

When I run " qmk setup", I got the "IndexError: list index out of range" with the same Error "UnicodeDecodeError: 'gbk' codec can't decode "

Hoping you may be able to help me with it

```
$ qmk setup
INFO Found qmk_firmware at C:/Users/GlaDo/qmk_firmware.

*** Would you like to set C:/Users/GlaDo/qmk_firmware as your QMK home? [y/n] y

INFO Wrote configuration to C:/msys64/mingw64/bin/qmk.exe.ini
INFO QMK Doctor is checking your environment.
INFO Detected Windows.
INFO All dependencies are installed.
INFO Found arm-none-eabi-gcc version 8.3.1
INFO Found avr-gcc version 8.3.0
INFO Found avrdude version 6.3
INFO Found dfu-util version 0.9
INFO Found dfu-programmer version 0.7.2
Exception in thread Thread-8:
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.8/threading.py", line 932, in _bootstrap_inner
    self.run()
  File "C:/msys64/mingw64/lib/python3.8/threading.py", line 870, in run
    self._target(*self._args, **self._kwargs)
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 1366, in _readerthread
    buffer.append(fh.read())
UnicodeDecodeError: 'gbk' codec can't decode byte 0x80 in position 15: illegal multibyte sequence
<class 'IndexError'>
ERROR list index out of range
Traceback (most recent call last):
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 635, in __call__
    return self.__call__()
  File "C:/msys64/mingw64/lib/python3.8/site-packages/milc.py", line 640, in __call__
    return self._entrypoint(self)
  File "C:/Users/GlaDo/qmk_firmware/lib/python/qmk/cli/doctor.py", line 302, in doctor
    sub_ok = check_submodules()
  File "C:/Users/GlaDo/qmk_firmware/lib/python/qmk/cli/doctor.py", line 124, in check_submodules
    for submodule in submodules.status().values():
  File "C:/Users/GlaDo/qmk_firmware/lib/python/qmk/submodules.py", line 21, in status
    git_cmd = subprocess.run(['git', 'submodule', 'status'], stdout=subprocess.PIPE, stderr=subprocess.PIPE, timeout=30, universal_newlines=True)
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 491, in run
    stdout, stderr = process.communicate(input, timeout=timeout)
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 1024, in communicate
    stdout, stderr = self._communicate(input, endtime, timeout)
  File "C:/msys64/mingw64/lib/python3.8/subprocess.py", line 1418, in _communicate
    stderr = stderr[0]
IndexError: list index out of range

```

zvecr, 2020-05-28:
> I download qmk_firmware for github (2020-5-27 09:23:51), unzip and rename to "qmk_firmware"

@Gladosxiao While not the cause of your issue, unfortunately thats not a valid workflow for QMK anymore. `qmk setup` will download the full repo including submodules which is required to get a working environment.

closed 2020-06-21 by Erovia