qmk/qmk_firmware #9027: Hasu USB2USB with AT90USB1286 firmware won't compile

duoshock, 2020-05-05:
I'm thinking of making a usb2usb with AT90USB1286, which allow me to have all the functions I need. Also, bigger EEPROM that allows me to have more than 1 layer with VIA enabled!

I tried to compile the firmware with 'MCU = AT90USB1286' in rule.mk and it won't compile, with error 'Please define board in avrpins.h' (see sceenshot)

Any pointers?

[image]

fauxpark, 2020-05-05:
The MCU name must be lowercase for AVRs.

duoshock, 2020-05-05:
> The MCU name must be lowercase for AVRs.

tried it. same error. :(

fauxpark, 2020-05-05:
Ah, you'll also need to add `#define CORE_TEENSY` in your config.h.

duoshock, 2020-05-05:
It works! thank you so much!

closed 2020-05-05 by duoshock