qmk/qmk_firmware #9039: [Bug]`KC_SLEP` & `KC_WAKE` are not recognized by MacOS Catalina.

s-show, 2020-05-07:
OS: MacOS Catalina 10.15.4

## Describe the Bug

In order to use `KC_SLEP` and `KC_WAKE`, these two key codes are assigned to the keyboard.

However, there is no response when I hold down either key.

I checked using an app that can capture typed keys (Karabiner-Eventviewer), but these two keys are not recognized.
When I quit Karanbiner-Elements, and hold these two keys, there is still no response.

Are these two keys no longer available due to the upgrade of MacOS?

## System Information

 - Keyboard:  Original Keyboard (I use Elite-C ver.3.1).
 - Operating system: MacOS Catalina 10.15.4
 - AVR GCC version: 8.4.0
 - ARM GCC version: 7.3.1 20180622 (release)
 - QMK Firmware version:0.8.7
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [x] Karabiner
   - [ ] Other:

## Additional Context

For your reference, here is a link to the FAQ page that explains the two key codes.

 [qmk_firmware/faq_keymap.md at master - qmk/qmk_firmware](https://github.com/qmk/qmk_firmware/blob/master/docs/faq_keymap.md)

fauxpark, 2020-05-07:
I'm also on 10.15.4 and do not see this. When I hold the keys for ~2 seconds this dialog appears:
[image]
Nothing happens on tap. This is normal behaviour for macOS. The media and power keys are not sent over the keyboard interface so it's unlikely Karabiner will pick them up.

Does the `KC_PWR` keycode also not work? If so, check that you have `EXTRAKEY_ENABLE = yes` in your rules.mk.

s-show, 2020-05-07:
When I checked `EXTRAKEY_ENABLE`, it was `no`. It now works properly after modifying `yes`.
Until then, `KC_POWER` was working properly, so I neglected to check it.
Thank you for your advice.

closed 2020-05-07 by s-show