qmk/qmk_firmware #9060: [Feature Request] Optional OLED displays

splitkb, 2020-05-09:
## Feature Request Type

- [ ] Core functionality
- [x] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Various (split) keyboards support OLED displays. These displays are often optional: users do not have to install them, as it functions as a keyboard just fine without.

Some users prefer to use the QMK Configurator. The Configurator does not provide support for OLED displays, but by means of keyboard default firmware, default support for OLED displays can still be added.

For the Kyria, I chose to enable the OLED by default so that every user can use it, even if they use the Configurator. However, this turned out to have a significant drawback: when a user does _not_ have an OLED installed, the keyboard becomes unresponsive.

I tried digging into the code but couldn't find the exact relevant piece. I suspect the firmware tries to initialize the display over I2C, but since there isn't a display, requests time out. This would happen on every matrix scan, causing many dropped keys - users report having keypresses sent only every two seconds.

I propose for the firmware to try to initialize the keyboard during startup, but to cease trying to initialize the keyboard after one or more attempts, as to prevent clogging up the matrix scan.

splitkb, 2020-05-09:
I spoke with @zvecr about this request and he offered the following code to be added to `<keyboard>.c`:

```c
void keyboard_pre_init_kb(void) {
  setPinInputHigh(D0);
  setPinInputHigh(D1);

  keyboard_pre_init_user();
}
```

This will not prevent the initialization mechanism from retrying, but it should prevent the I2C timeout from being hit. The code above sets the I2C pins to high, causing them to be set to a voltage instead of floating.

The request still stands, the code above is a potential workaround.

splitkb, 2020-12-04:
I'll want to resolve this issue on my next keyboard since I'm planning to allow for socketable displays, so I'm commenting to keep the issue open. Thanks!

mrlinuxfish, 2020-12-10:
Is it possible to implement a timeout so if the controller doesn't detect a display it will stop trying to write data to the display?