qmk/qmk_firmware #9064: [Potential Bug] With light layers enabled, changing RGB animations doesn't allow you to preview

nblyumberg, 2020-05-10:
## Describe the Bug
With light layers enabled, cycling through the animations doesn't allow a preview. Basically my setup is to have RGB_MOD or RGB_RMOD on a non default layer so to change the animations I am always at a higher level which has a different color. Cycling through the animations doesn't allow a preview as the light layer behavior appears to override the preview. I think a viable workaround on this issue is to have RGB_MOD and RGB_RMOD start a  timer that suspends the light layer behavior so that animations can be previewed if they are being changed and RGB_MOD/RGB_RMOD keycodes are seen 

## System Information

 - Keyboard:  Super16
   - Revision (if applicable):  
 - Operating system: Linux
 - AVR GCC version: 5.4.0

 - ARM GCC version: arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620

 - QMK Firmware version: 0.8.148
 

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

spidey3, 2020-05-20:
This is an interesting thought, but only an issue if the lighting layers that are currently active cover the entire set of LEDs.

I suppose that one potential solution would be to _pause_ the lighting layers for a set number of milliseconds after the mode / hue / sat / val is changed. But that would add rather a bit of complexity, and take yet more flash & RAM...

nblyumberg, 2020-05-24:
I've played around with light layers today more and got the keymap closer to how I want it to act but the lack of preview with light layers overriding everything is a real killjoy :/
My current layout is 
Layer 0 - eeprom reset to orange or can be dialed in with RGB controls
Layer 1 - Green (Alt+F13 to F24) 
Layer 2 - Red (Shift+F13 to F24)
Layer 3 - Blue (Ctrl+F13 to F24)
Layer 4 - White (Multimedia layer)
Layer 5 - RGB adjustments (But with light layers this becomes unusable)

I'm sure this is also complex (if not impossible) but what about either having the light layers on/off functions or defining a layer on which light layer doesn't run?

spidey3, 2020-05-24:
This can be implemented without a change to core. 
Here is code that you can add to your userspace that should give you a 3 second "preview":
```c
#if defined(RGBLIGHT_ENABLE) && defined(RGBLIGHT_LAYERS)
static uint32_t rgb_preview_timer = 0;
#endif

void matrix_scan_user(void) {
# if defined(RGBLIGHT_ENABLE) && defined(RGBLIGHT_LAYERS)
    if (rgb_preview_timer && TIMER_DIFF_32(timer_read32(), rgb_preview_timer) > 3000) {
        rgb_preview_timer = 0; 
        default_layer_state_set_user(default_layer_state);
        layer_state_set_user(layer_state);
        led_update_user((led_t) host_keyboard_leds());
    }
# endif
}

void post_process_record_user(uint16_t keycode, keyrecord_t *record) {
# if defined(RGBLIGHT_ENABLE) && defined(RGBLIGHT_LAYERS)
  switch (keycode) {
    case RGB_TOG ... VLK_TOG:
      for (uint8_t i = 0; i < RGBLIGHT_MAX_LAYERS; i++) {
        rgblight_set_layer_state(i, false);
      }
      rgb_preview_timer = timer_read32();
      break;
  }
# endif
  return;
}
```

nblyumberg, 2020-05-24:
This worked beautifully! I fully support having this added to core as it fixes what I think is a design issue with light layers.

spidey3, 2020-05-24:
Because the logic to set lighting layers on and off is currently in userspace, the preview probably should also be in userspace. But if we promote the implementation of keymap layer indication and host led indication using lighting layers to core, then at that time we should also promote the preview function.

For example, there are a bunch of assumptions made here about how lighting layers are used. What if there is other static state being indicated using lighting layers besides the keymap layer state and host led state? This code (without changes) would reset those lighting layers, and not properly reset them after the preview...

closed 2022-08-28 by zvecr