qmk/qmk_firmware #9078: [Bug] preonic/rev3 fails to flash with monotone sound

dyamon, 2020-05-12:
## Describe the Bug

Flashing Preonic/rev3 with default or custom keymaps succeeds but the keyboard is unresponsive and emitting a monotone continuous sound (instead of the standard three-tone connection sound).

## System Information

 - Keyboard: Preonic
   - Revision (if applicable): rev3
 - Operating system: Gentoo Linux (5.4.38-gentoo-x86_64)
 - AVR GCC version: avr-gcc (Gentoo 8.4.0 p1) 8.4.0
 - ARM GCC version: arm-none-eabi-gcc (Gentoo 8.4.0 p1) 8.4.0
 - QMK Firmware version: 0.8.147
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [x] `setxkbmap` (removes CAPS-LOCK, activated pressing both shift keys instead)
   - [x] `xcape` (CAPS-LOCK acts as Control-L and Esc)
   - [x] `sxhkd` handles keymaps at Window Manager level

## Additional Context

Flashing and compilation is executed with the following commands:
```
make preonic/rev3:<keymap> EXTRAFLAGS+=--specs=nosys.specs
make preonic/rev3:<keymap>:dfu-util EXTRAFLAGS+=--specs=nosys.specs
```
The reason for `EXTRAFLAGS+=--specs=nosys.specs` is explained in this issue #8809 

The keyboard is in flashing mode and the flashing process succeeds with this output:
```
QMK Firmware 0.8.147
Making preonic/rev3 with keymap default and target dfu-util

arm-none-eabi-gcc (Gentoo 8.4.0 p1) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/audio/muse.c                                                                     [OK]
Compiling: keyboards/preonic/preonic.c                                                              [OK]
Compiling: keyboards/preonic/rev3/rev3.c                                                            [OK]
Compiling: keyboards/preonic/keymaps/default/keymap.c                                               [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: quantum/process_keycode/process_audio.c                                                  [OK]
Compiling: quantum/process_keycode/process_clicky.c                                                 [OK]
Compiling: quantum/audio/audio_chibios.c                                                            [OK]
Compiling: quantum/audio/voices.c                                                                   [OK]
Compiling: quantum/audio/luts.c                                                                     [OK]
Compiling: quantum/process_keycode/process_music.c                                                  [OK]
Compiling: tmk_core/common/chibios/eeprom_stm32.c                                                   [OK]
Compiling: tmk_core/common/chibios/flash_stm32.c                                                    [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgblight.c                                                                       [OK]
Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
Compiling: drivers/chibios/ws2812.c                                                                 [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: quantum/encoder.c                                                                        [OK]
Compiling: quantum/dip_switch.c                                                                     [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/crt0_v7m.S                           [OK]
Assembling: lib/chibios/os/common/startup/ARMCMx/compilers/GCC/vectors.S                            [OK]
Assembling: lib/chibios/os/common/ports/ARMCMx/compilers/GCC/chcoreasm_v7m.S                        [OK]
Compiling: tmk_core/common/host.c                                                                   [OK]
[... 179 lines cut ...]
[18182.783941] usb 1-1.1.4.1: Manufacturer: OLKB
[18182.783944] usb 1-1.1.4.1: SerialNumber: 0
[18182.786678] input: OLKB Preonic as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4.1/1-1.1.4.1:1.0/0003:FEED:6061.0021/input/input68
[18182.841783] hid-generic 0003:FEED:6061.0021: input,hidraw0: USB HID v1.11 Keyboard [OLKB Preonic] on usb-0000:00:1a.0-1.1.4.1/input0
[18182.844942] input: OLKB Preonic Mouse as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4.1/1-1.1.4.1:1.1/0003:FEED:6061.0022/input/input69
[18182.845370] input: OLKB Preonic System Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4.1/1-1.1.4.1:1.1/0003:FEED:6061.0022/input/input70
[18182.901433] input: OLKB Preonic Consumer Control as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4.1/1-1.1.4.1:1.1/0003:FEED:6061.0022/input/input71
[18182.901749] input: OLKB Preonic Keyboard as /devices/pci0000:00/0000:00:1a.0/usb1/1-1/1-1.1/1-1.1.4/1-1.1.4.1/1-1.1.4.1:1.1/0003:FEED:6061.0022/input/input72
[18182.902296] hid-generic 0003:FEED:6061.0022: input,hidraw1: USB HID v1.11 Mouse [OLKB Preonic] on usb-0000:00:1a.0-1.1.4.1/input1
[18182.903989] hid-generic 0003:FEED:6061.0023: hiddev96,hidraw2: USB HID v1.11 Device [OLKB Preonic] on usb-0000:00:1a.0-1.1.4.1/input2
```

The output of `qmk doctor` is
```
Ψ QMK Doctor is checking your environment.
Ψ Detected Linux.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.4.0
Ψ Found avr-gcc version 8.4.0
Ψ Found avrdude version 6.3
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.7.2
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

chandrahmuki, 2020-11-30:
same problem here ..did you find anything related to that bug ?

dyamon, 2020-11-30:
I did not. I used the other (working) machine instead. Still I would like to solve this problem. Do you have the same exact settings and outputs I have? Maybe we can narrow down the problem.

chandrahmuki, 2020-11-30:
i have exactly the same output i had to modify the chibios.mk and switch to softfp/ soft in the Flags. thats the first thing because i had a bunch of VFP errors without it then i added the extraflags like you did.
i tried also to update my arm compiler i was using the default 8.4 frozen on the sound. then i upgrade to 9.3 also frozen on sound but this time with the red led activated then i tried to upgrade to 10.2 and this time it was on frozen but no sound anymore and still the red led active ...so there is something around the compilation ...i suppose.

dyamon, 2021-01-04:
I still have no clue about this, at least with the current information. Maybe someone knows how to obtain more diagnostics from the keyboard and/or the compilation process.

drashna, 2024-10-13:
I believe this has been fixed for a while now, due to some fixes to the audio drivers.

closed 2025-02-10 by github-actions[bot]