qmk/qmk_firmware #9085: [Bug] Calling lcd_clrsrc() from keyboard.c raises error

RichardMiruGeorge, 2020-05-13:
Trying to interface my LCD with QMK with HD44780_ENABLE = yes in the rules.mk raises following issue:
The following lines in my keyboard.c:

void matrix_init_kb(void) {
    // put your keyboard start-up code here
    // runs once when the firmware starts up

    matrix_init_user();
    lcd_clrsrc();
//    lcd_init(LCD_DISP_ON_CURSOR_BLINK);
}

cause the error:
-----------------------------------------------------
qmk_firmware$ make kb:default
QMK Firmware 0.8.153
Making kb with keymap default

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  22382	      0	  22382	   576e	.build/kb_default.hex

Compiling: keyboards/kb/kb.c                                                                       keyboards/kb/kb.c: In function ‘matrix_init_kb’:
keyboards/kb/kb.c:10:5: error: implicit declaration of function ‘lcd_clrsrc’ [-Werror=implicit-function-declaration]
     lcd_clrsrc();
     ^
cc1: all warnings being treated as errors
 [ERRORS]
 | 
 | 
 | 
tmk_core/rules.mk:383: recipe for target '.build/obj_kb_default/keyboards/kb/kb.o' failed
make[1]: *** [.build/obj_kb_default/keyboards/kb/kb.o] Error 1
Make finished with errors
Makefile:560: recipe for target 'kb:default' failed
make: *** [kb:default] Error 1

fauxpark, 2020-05-13:
Looks like the documentation is wrong, the function name is actually `lcd_clrscr()`. If that still doesn't work, try adding `#include "hd44780.h"` at the top of the keyboard.c.

RichardMiruGeorge, 2020-05-13:
ok now i feel silly - that works! - thanks for doing the typo spotting for me.
For posterity: #include "hd44780.h" is unnecessary as its included in  tmk_core/common/keyboard.c when the appropriate enable is set in rules.mk

PR #9086 (2020-05-13): fixed typo for clear lcd screen function in hd44780 docs
brought to attention by RichardMiruGeorge and fauxpark in #9085

## Description

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [ ] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [x] Documentation

## Issues Fixed or Closed by This PR

* fixes #9085

## Checklist

- [ ] My code follows the code style of this project.
- [ ] My change requires a change to the documentation.
- [ ] I have updated the documentation accordingly.
- [x] I have read the [**CONTRI [... cut]
closed 2020-05-13 by RichardMiruGeorge