qmk/qmk_firmware #9104: [Bug] Board with atmega328p fails to compile when MIDI is enabled

mavanmanen, 2020-05-15:
## Describe the Bug
I'm trying to write a keymap for the Plaid Pad by Keycapsss which uses an atmega328p.
When trying to compile the firmware I receive the following error:
```
Compiling: keyboards/keycapsss/plaid_pad/plaid_pad.c
In file included from tmk_core/common/host.h:23,
                 from quantum/keymap.h:33,
                 from quantum/quantum.h:30,
                 from keyboards/keycapsss/plaid_pad/plaid_pad.h:3,
                 from keyboards/keycapsss/plaid_pad/plaid_pad.c:1:
tmk_core/common/host_driver.h:24:14: fatal error: midi.h: No such file or directory
 #    include "midi.h"
              ^~~~~~~~
compilation terminated.
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:386: .build/obj_keycapsss_plaid_pad_mavanmanen/keyboards/keycapsss/plaid_pad/plaid_pad.o] Error 1
Make finished with errors
make: *** [Makefile:582: keycapsss/plaid_pad:mavanmanen] Error 1
```

However, when building firmware for a different board with the atmega32u4 it works without any problems.

## System Information
 - Keyboard:  Plaid Pad by Keycapsss
 - Operating system: Windows 10
 - AVR GCC version: 8.3.0
 - QMK Firmware version: 0.8.160

mavanmanen, 2020-05-15:
Here is my code as well: https://github.com/mavanmanen/qmk_firmware/tree/master/keyboards/keycapsss/plaid_pad/keymaps/mavanmanen

fauxpark, 2020-05-15:
MIDI is not currently supported on boards running V-USB. Due to the technical limitations of doing USB in software, it might never be. It uses a bulk-in endpoint and a bulk-out endpoint, so you would probably have to disable everything but the keyboard endpoint, as is done now for VIA.

mavanmanen, 2020-05-15:
That's interesting. I've seen MIDI projects using the atmega328p + V-USB before though. Is it just that it can't work in conjunction with QMK?

fauxpark, 2020-05-15:
At the moment it's simply that no one has implemented it yet - personally I would like to wait for consolidation of the USB descriptor stuff to include V-USB (right now only ChibiOS and LUFA share their descriptors) before improving cross-platform support for features like this.

It does seem doable, but as I mentioned, you'd need to sacrifice VIA, media keys, pretty much everything that adds an additional endpoint descriptor except for the normal 6KRO keyboard interface; you only get two IN (device-to-host) endpoints with V-USB.

mavanmanen, 2020-05-15:
I guess we'll have to wait and see then. I suppose if I wanted to run the pad as just a MIDI controller I'd now have to implement everything myself yes?

fauxpark, 2020-05-15:
I'd say so. The MIDI support is kind of coupled to the LUFA framework, so you'd have to rewrite it a little to be more generic.

jonmccon, 2020-05-21:
I'm trying to build a similar midi & keycode device as well. However getting a similar compile issue on a Proton C which is an STM32. 

```
Compiling: tmk_core/protocol/chibios/main.c                                                        In file included from tmk_core/protocol/chibios/main.c:51:
tmk_core/protocol/midi/qmk_midi.h:7:36: error: unknown type name 'MIDI_EventPacket_t'
 void              send_midi_packet(MIDI_EventPacket_t* event);
                                    ^~~~~~~~~~~~~~~~~~
tmk_core/protocol/midi/qmk_midi.h:8:36: error: unknown type name 'MIDI_EventPacket_t'
 bool              recv_midi_packet(MIDI_EventPacket_t* const event);
                                    ^~~~~~~~~~~~~~~~~~
```

mavanmanen, 2020-05-22:
For now I've found a workaround using raw HID messages and a driver in csharp. https://github.com/mavanmanen/Plaid-Pad-MIDI

xbong360, 2020-06-16:
> I'm trying to build a similar midi & keycode device as well. However getting a similar compile issue on a Proton C which is an STM32.
>
> ```
> Compiling: [...]

Did this ever get figured out? I have midi layers on a BDN9 that work great, but I can't get midi parameters working on the rotary encoders due to the same compile issue from including qmk_midi.h.

closed 2022-08-28 by zvecr