qmk/qmk_firmware #9120: Building qmk firmware with custom compiler and linker flags

heshananupama, 2020-05-17:
Hi. I had set up qmk development using eclipse by following the guide at https://docs.qmk.fm/#/other_eclipse. For my research work I needed to build firmware with different compiler flags. For examples I need to build the same firmware with different optimization levels and different debug levels. However, when I go to properties and build settings in eclipse there are no ways to change the build settings by passing different flags to compiler and the linker. Is there a way that I could change these flags. Any help would be much appreciated thanks

elfmimi, 2020-05-22:
QMK is complex but relatively well organized.

You can find the variable related to optimization level here.
https://github.com/qmk/qmk_firmware/blob/38d2c8cedf9f9f43b5ba372ffd38ccc7cc96be15/tmk_core/rules.mk#L49

And here is another one related to link-time-optimization, or lto  if you like.
https://github.com/qmk/qmk_firmware/blob/38d2c8cedf9f9f43b5ba372ffd38ccc7cc96be15/tmk_core/common.mk#L156

In a use case like you described, I guess it is better to use make command directly.
for eample `make planck/rev6:default OPT=2 LINK_TIME_OPTIMIZATION_ENABLE=yes`

and you can confirm what was the actual compiler flags used by looking into the following files.
.build/obj_planck_rev6_default/cflags.txt
.build/obj_planck_rev6_default/cxxflags.txt
.build/obj_planck_rev6_default/asflags.txt
.build/obj_planck_rev6_default/ldflags.txt

heshananupama, 2020-05-22:
Thank you very much!

closed 2020-05-22 by heshananupama