qmk/qmk_firmware #9123: [YMDK b.face] RGB support?

shohart, 2020-05-17:
Hello, there, fellow, keebpeople. today i fainally decided to move my old YMDK b.face board to QMK firmware, because it is much more awesome, than ps2avrGB, that it is shipped with. 
I managed to make a keymap, that fit my needs, but i realized that RGB support is not implemented in this particular board yet.

So the quastion is. Is it possible to implement RGB lights in this board? If so, how can one do it. 
I am not that expirienced in qmk coding, but i assume that one can accommodate files from similar boards like bfake for example... 

So any help will be apreciated.
Thanks

fauxpark, 2020-05-17:
Given this is a PS2AVRGB board, it should be as simple as setting `RGBLIGHT_ENABLE = yes` and `WS2812_DRIVER = i2c` in rules.mk, and adding the necessary config stuff in config.h.

shohart, 2020-05-17:
Such an easy solution! 
i completelly forgot about the rules.mk! I tried messing around with all other files with no luck at all. Thanx m8!

P.S. mine has onboard RGB LEDs, but they seem to be of very bad quality or maybe the board itself is the problem, because i had to replace 2 of them as they stopped to work. For some reason now one of the chain (latest) does not operate even when i replace it with working one. maybe some schematics issue... :(

fauxpark, 2020-05-17:
Did you set `RGBLED_NUM` correctly in config.h?

shohart, 2020-05-17:
> Did you set `RGBLED_NUM` correctly in config.h?

yeah i did - it has 12 LEDs so i set it to 12, i tried to input 13 and 14 - same result. setting it to 11 make 2 LEDs go dark though -)

fauxpark, 2020-05-18:
In that case I'd say there might be a soldering issue? Check for continuity with VCC, GND and DO of the previous LED to DI of the nonworking one. Since it's the last in the chain, DO will not be connected to anything.

shohart, 2020-05-18:
Thanx m8 i will check it then! it would be nice to have board schematics to check it properly, but thanx for the hint! 
Maybe it was first faulty LED from the very begining and i fried next two, that i tried to replace because used my soldering iron instead of hot air blower because I was too lazy to get it from the shelf ¯\_(ツ)_/¯. Gonna close this issue now because you solved the main question. Thanx again!

closed 2020-05-18 by shohart

fauxpark, 2020-05-18:
The circuit for PS2AVRGB boards is pretty much standardised and can be found here: https://github.com/showjean/ps2avrU/blob/master/firmware/ps2avrGB/Circuit/ps2avrGB/20140829_140848_001.png