qmk/qmk_firmware #9131: Virtual serial is not documented

iFreilicht, 2020-05-18:
We have a virtual serial port available since 2016 (#677), and the stenography features are using it, yet it doesn't seem to be documented anywhere and only [one keymap is using the virtser_recv](https://github.com/qmk/qmk_firmware/blob/master/users/miles2go/keymaps/handwired/ms_sculpt_mobile/keymap.c) for receiving commands from the PC, something that seems to be asked about every now and then.

It seems it's as easy as adding `VIRTSER_ENABLE = yes` to `rules.mk`, a line that is not present in any example file, and then defining `virtser_recv` in your `keymap.c`.

If I got the structure of the docs right, this new page should go into the "Software Features" section.

drashna, 2020-05-20:
Yup to just about all of that.  The send/receive stuff should work very much like how the raw HID does.  So a lot of the code should be convertible.

stapelberg, 2020-11-21:
@Stale: this issue is still relevant

danshick, 2021-06-02:
@Stale This issue is still relevant.

zvecr, 2022-08-29:
This issue has been automatically closed because it has not had any recent activity.
If this issue is still valid, re-open the issue and let us know.

closed 2022-08-29 by zvecr

iFreilicht, 2022-08-30:
@zvecr This issue is not fixed yet and it is still active. The missing documentation has not been written.

keyboard-magpie, 2022-08-30:
> […](#)

If it's not happened in 2 years, is it going to happen at all? Issues shouldn't remain open indefinitely and this was closed as part of a cleanup.

PeterHindes, 2023-01-09:
I'll see if I can figure out how it works. No promises though.