qmk/qmk_firmware #9138: Add KBDFans Tofu96

McCroden, 2020-05-19:
The [KBDFans Tofu96](https://kbdfans.com/collections/group-buy/products/tofu96-mechanical-keyboard-diy-kit) has shipped and is not available yet in qmk configurator, which uses this repo (I opened an issue with qmk configurator first). 

I asked KBDFans for a link to the firmware and was provided:
This url: http://www.qmkeyboard.cn/
Choose "DZ96" for the appropriate qmk firmware.

I'm new to keyboards and qmk. I'm not sure what the correct process is to get this added to qmk. I read the contributing docs but didn't quite understand if this was the correct process or not.

McCroden, 2020-05-20:
Wanted to add some information. The [Espectro](https://config.qmk.fm/#/espectro/LAYOUT_split_shift_and_bs) worked. It was recommended [from my issue on qmk configurator](https://github.com/qmk/qmk_configurator/issues/734).

JosephAHK, 2020-09-08:
I wanted to add that I was able to go to the [chineese QMK configurator website](www.qmkeyboard.cn) and download the DZ96 source code and compile it locally using an updated QMK version. The QMK version in the HEX files produced by the outdated site is one from 2017. If we could get the dz96 added to the official source that would be awesome.

Jon-Muller, 2020-11-03:
Could you please elaborate? I am a complete noob, and while I was able to get a .hex file download for my DZ96 I am unsure how to flash my board as it is not listed in the QMK Toolbox.

JosephAHK, 2020-11-03:
Go to https://config.qmk.fm/#/espectro/LAYOUT_default and compile a hex
using the espectro keymap instead

Jon-Muller, 2020-11-05:
I ended up reading this elsewhere as well. Thank you, it worked perfectly!

closed 2021-03-16 by stale[bot]