qmk/qmk_firmware #9172: Help with compiling Planck keymap

GrinninBarrett, 2020-05-23:
I decided to make some changes to my Planck layout, and I did so by simply opening the old keymap.c file I created using the QMK docs last summer sometime. I haven't used QMK much since then, other than the configurator for other keyboards, which is its own thing. After editing the keymap.c file, which was still in its proper location according to the docs (in keyboards>Planck>keymaps>grinninbarrett), I attempted to run the `make` command as I had done in the past, and it didn't work. I checked the QMK docs which seem to have been overhauled since I last tried to compile a firmware, and it seemed I needed to update Homebrew and/or QMK, as the `qmk` command was not a valid command according to iTerm. I did all of that, and although it now recognizes the `qmk` command, it won't compile, even though I believe I'm doing everything according to the docs. 

I've attached a screenshot of my CLI after trying to use these new commands. What have I done wrong, and how can I compile that keymap? What other information can I give you to help with diagnosing this problem? 

Thank you!

[image]

GrinninBarrett, 2020-05-23:
Also, I am trying to create a keycode that will send Caps lock when tapped and MO to a new layer when held. I have defined that layer with `#define NUMPAD MO(_NUMPAD)`. Would the correct code for that be `MT(NUMPAD, KC_CAPS)`. I do apologize for my grave lack of understanding when it comes to programming keyboards! Thank you for your help!

nblyumberg, 2020-05-25:
Hello,

Do you know the Planck revision you have?  It looks like you are omitting the rev folder so it's failing to compile.

You would want to do something like this in the qmk directory:
`qmk compile -kb planck/rev6 -km grinninbarrett `

The options in that folder are:
rev1 to rev6

For your keycode question, I don't believe you can do what you want with MT. You'll likely can use LT for that, where LT(1, KC_CAPS) as your KC_CAPS replacement on the Planck.

GrinninBarrett, 2020-05-26:
Hi! Thanks for the reply. I did try to include rev6 in exactly the way you
mentioned, but it also didn’t work. It seems that the “compile” command is
what is not working. This is the recent release from Drop, but I have
another, EOTW rev6 Planck I got last year from OLKB that I didn’t have any
problem with at the time, but haven’t tried to flash recently. There is
something not working about the QMK functionality from the command line,
and I don’t know enough about git or the command line in general to
understand what the problem could be. I’ve downloaded the most recent
version of everything, I think. I’m not sure why the compile command won’t
work. It says to try either clone or setup. I did both - clone just clones
successfully (no idea to where, or what that does), and setup only asks one
y/n question and stops. It doesn’t seem that I can move further. Any ideas?

nblyumberg, 2020-05-26:
Do you have any screenshots of what you encountered? 

This definitely works as I tried compiling the default keymap earlier:

```
% qmk compile -kb planck/rev6 -km default
Ψ Compiling keymap with make planck/rev6:default

QMK Firmware 0.8.110
Making planck/rev6 with keymap default

arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: quantum/audio/muse.c                                                                     [OK]
Compiling: keyboards/planck/planck.c                                                                [OK]
Compiling: keyboards/planck/rev6/rev6.c                                                             [OK]
Compiling: keyboards/planck/keymaps/default/keymap.c                                                [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
.
.
.
Linking: .build/planck_rev6_default.elf                                                             [OK]
Creating binary load file for flashing: .build/planck_rev6_default.bin                              [OK]
Creating load file for flashing: .build/planck_rev6_default.hex                                     [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  56164	      0	  56164	   db64	.build/planck_rev6_default.hex

Copying planck_rev6_default.bin to qmk_firmware folder                                              [OK]
(Firmware size check does not yet support cortex-m4 microprocessors; skipping.)

```

GrinninBarrett, 2020-05-26:
I will have to try tomorrow morning now but I will get back to you as soon
as I try it again.

nblyumberg, 2020-05-26:
Okay, also try `qmk doctor` and `qmk --version` to give extra info.

nblyumberg, 2020-06-02:
Were you able to resolve this issue?

GrinninBarrett, 2020-06-02:
Hello,

No, not yet. I tried compiling the default keymap in the way you mentioned,
and I'm receiving the same errors. I also tried qmk doctor, which gave the
same errors as well. Do you have any suggestions for this?

Thank you,
Tucker Barrett

nblyumberg, 2020-06-03:
Ohhhh...I think i see it. In your screenshot it's telling you that valid options are clone and setup. So you likely didn't run the [setup part](https://beta.docs.qmk.fm/tutorial/newbs_getting_started#3-run-qmk-setup-id-set-up-qmk) of getting started.

Try this: `qmk setup` and then you should be able to compile.

GrinninBarrett, 2020-06-03:
I have run that multiple times, but it only asks one question. I've
attached a screenshot. It's the same thing it's done every time I've run
it. I also ran "clone" for that reason, and it just gave some output and
didn't seem to actually affect anything. It continues to say qmk compile is
not recognized. I'm really confused about it. I'm on the verge of
formatting my computer to get rid of iTerm, as well as to re-download QMK
stuff, to see if that helps. Any further advice is definitely appreciated.

Tucker

nblyumberg, 2020-06-05:
I think if you can stop by the QMK Discord someone there may be able to get you further. I don't see your updated screenshot but the original one in your first post shows that you try to run a compile command:
`qmk compile planck -km grinninbarrett`

and QMK complains telling you that your two *valid* options are to `clone` or `setup`

You would have to paste more terminal output after you type in `qmk setup` so we can see if that gives more clues on what we need.

GrinninBarrett, 2020-06-05:
Yes that’s the screenshot I included before. I’ll send again, maybe it will
show this time.

nblyumberg, 2020-06-09:
Apparently GitHub comments do not allow image attachments. You would need to copy and paste all the messages you are seeing including the commands that you are running. This should give us more ideas.

GrinninBarrett, 2020-06-09:
So, the following is the output following the qmk setup command after
cd-ing to qmk_firmware:

qmk setup
Ψ Found qmk_firmware at /Users/ctbarrett/qmk_firmware.

*** Would you like to set /Users/ctbarrett/qmk_firmware as your QMK home?
[y/n] y

Ψ Wrote configuration to '/Users/ctbarrett/Library/Application
Support/qmk/qmk.ini'
ψ QMK Doctor is Checking your environment
ψ Detected macOS
ψ QMK is ready to go

But then, after all of that, it still will not allow me to run "qmk
compile". Any ideas?

Thank you,
Tucker

nblyumberg, 2020-06-10:
Can you also provide the output of qmk compile right after? Also if you're able to join the QMK discord we can try to give you faster help.

closed 2020-10-08 by stale[bot]