qmk/qmk_firmware #9204: [Feature Request] Swap buttons on PS2 Mouse/Trackball

jkutianski, 2020-05-26:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

At the moment doesn't exist any way to swap left and rigth button on PS2 mouse connected via QMK. 
The Logitech T-SBC-12 trackball integrated on the Compaq MX-11800 has 2 buttons the top is the left button and the botton is the rigth one. For the people that suffers for carpal tunnel syndrome this is very bad because you are forced to use the middle finger in a very unconfortable position for drag and drop. Swapping the buttons solve this problem.

tyetye, 2020-06-03:
Can you use these defines to accomplish this?
```
#define PS2_MOUSE_BTN_LEFT      0
#define PS2_MOUSE_BTN_RIGHT     1
#define PS2_MOUSE_BTN_MIDDLE    2

```
They're documented in qmk's [PS/2 Mouse pages](https://beta.docs.qmk.fm/using-qmk/hardware-features/feature_ps2_mouse).

jkutianski, 2020-06-04:
> Can you use these defines to accomplish this?
> 
> ```
> #define PS2_MOUSE_BTN_LEFT      0
> #define PS2_MOUSE_BTN_RIGHT     1
> #define PS2_MOUSE_BTN_MIDDLE    2
> ```
> 
> They're documented in qmk's [PS/2 Mouse pages](https://beta.docs.qmk.fm/using-qmk/hardware-features/feature_ps2_mouse).

Nope. If you change these don't affect anything
Check it here https://github.com/qmk/qmk_firmware/blob/master/tmk_core/protocol/ps2_mouse.c

tyetye, 2020-06-04:
Ooh, interesting. Thank you for pointing that out. I was planning to use those in my own project.

closed 2020-10-03 by stale[bot]