qmk/qmk_firmware #9223: [Feature Request] ESP32 and/or ESP8266 support?

jmding8, 2020-05-27:
## Feature Request Type

- [X] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

There are many inexpensive ESP32 and ESP8266 based dev boards with on-board bluetooth, li-po charging, wifi, and sometimes OLED screens. These seem like a great upgrade from our usual Teensy and Pro Micro dev boards given how feature rich they are, and the fact that they are even cheaper in a lot of cases than the Pro Micro.

[Example 1](https://www.aliexpress.com/item/4000340140634.html?spm=a2g0o.productlist.0.0.3a2640d9HwEboB&algo_pvid=b825ad22-c20e-4164-a50f-f034f0294be0&algo_expid=b825ad22-c20e-4164-a50f-f034f0294be0-1&btsid=0be3764515906087051605230eea01&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)

[Example 2](https://www.aliexpress.com/item/4000253063093.html?spm=a2g0o.productlist.0.0.3a2640d9HwEboB&algo_pvid=b825ad22-c20e-4164-a50f-f034f0294be0&algo_expid=b825ad22-c20e-4164-a50f-f034f0294be0-23&btsid=0be3764515906087051605230eea01&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_)

I was pretty surprised to not find any discussion about these dev boards in the keyboard community. Would it be possible to get official QMK support for these dev boards? Or are there some technical reasons why it would not be possible?

MxBlu, 2020-05-27:
I remember looking at this a few years back, one of the reasons is because the ESP8266's USB port is actually going through a UART to USB bridge, so you can't use it to enumerate as a HID device and send HID reports. You can, however, implement something like V-USB if you put your mind to it, but that would require using other pins and wiring your own usb cable/port from it.

Basically, it's not ideal. They're good IoT boards, but I don't see them being too useful as a USB keyboard.

Also regarding the bluetooth stack, I've seen work on a nicer approach to managing the bluetooth stack in QMK. Not sure how things will proceed from there, but that approach might work if that is the case.

fauxpark, 2020-05-28:
The ESP chips run on the Xtensa architecture, which is completely different to the platforms we currently support (ARM and AVR). It would likely be a significant effort to add this entirely new one given that ChibiOS almost certainly would not accept it. Most importantly though, whoever does this would then be responsible for maintaining it.

uyoyo117, 2020-08-01:
maybe add UART sending of key code, this way any board can simply be used as a bridge

eroldan, 2020-10-22:
ESP32-S2 supports USB https://www.espressif.com/en/news/espressif-announces-%E2%80%A8esp32-s2-secure-wi-fi-mcu

tzarc, 2020-10-22:
This should cover what's required, specifically the section _"Adding support for a new MCU Family"_: [Selecting an Arm MCU](https://docs.qmk.fm/#/platformdev_selecting_arm_mcu?id=add-new-mcu-family)

christiangda, 2020-12-07:
Could be great use 1 or 2 of these 

https://heltec.org/project/wifi-kit-32/ 

integrated with the custom keyboards. 

* Bluetooth
* WIFI
* OLED
* Battery charger and connector 
* Reset button
* More Memory and CPU!
* etc

all of these only $9

JesusFreke, 2021-03-24:
I was poking at this a bit today and managed to get something.. well. not working. but at least compiling and linking. You can follow along over at https://hackaday.io/project/178232-lalboard/log/190758-investigating-qmk-on-the-esp32 if you're interested.

tzarc, 2021-03-24:
Given that esp32 is supported by RIOT-OS, perhaps that would be worth investigating as a HAL?
At this stage we don't have the bandwidth to support or maintain HAL libraries for just QMK, which is why we've been contemplating integrating established projects like RIOT-OS or TinyUSB to provide extra hardware support. Ultimately, we can rely on upstream maintenance instead of letting things bit-rot like the current ATSAMD51 support for the Massdrop boards.

bilogic, 2021-07-20:
@JesusFreke 

> Given that esp32 is supported by RIOT-OS, perhaps that would be worth investigating as a HAL?
> At this stage we don't have the bandwidth to support or maintain HAL libraries for just QMK, which is [...]

Do you have a fork or something? I have the Logitech RF protocol working on the ESP8266 https://github.com/bilogic/logitech-unifying-device

JesusFreke, 2021-07-31:
Yeah, I have a fork that's integrated with esp-idf, not RIOT-OS. But I don't have it published or anything yet. And It's not really an approach that would be acceptable to merge into mainline qmk. It basically builds qmk as a esp-idf module, and uses esp-idf to build the whole project.

helman, 2021-08-02:
@JesusFreke do you mind sharing your work, i'm currently building dactyl manuform, and thinking about to add a bluetooth support.
I saw a mitosis hardware that is leveraging ble module for it split kb, but not sure if that would be possible to modify it as the main bl connection to devices. ;)

bilogic, 2021-08-02:
@JesusFreke yea, I'm keen to see what it can do too!

jorgeluisrezende, 2022-03-08:
any news about it?

Turegano, 2022-05-30:
It's interesting to have alternatives due IC shortage

oelburk, 2022-08-19:
Any updates on this?

keyboard-magpie, 2022-08-19:
zvecr has made significant progress with RIOT support- not with this in mind though. 

I'd say it's still a good way away from being at the point where you could be considering adding ESP32...

tzarc, 2022-08-29:
At this stage asking for status updates is a hindrance rather than a help -- closing this issue until someone either provides an implementation for review, or RIOT gets to the point where this becomes feasible.

closed 2022-08-29 by tzarc

PeterHindes, 2022-12-18:
I'm wondering how hard it would be to get USB keyboard functionality working on an esp32-s3. That could be a good starting point to start delving deeper into the code.

EntropyValley, 2023-08-29:
I may be wrong, but looking through the RIOT docs, it seems like the ESP32 is supported at this point. https://api.riot-os.org/group__cpu__esp32.html

wdtamagi, 2023-09-09:
Is there any guide how to use RIOT with the QMK?