qmk/qmk_firmware #9230: [Bug] `RETRO_TAPPING` does not work with `ALT_T(*)`

tom-tan, 2020-05-29:
## Describe the Bug

Even when `RETRO_TAPPING` is defined, `ALT_T(*)` works as same as without `RETRO_TAPPING`.

I reproduced this issue as follows:
- Add the following to `crkbd/keymaps/default/config.h`:
   ```c
   #define RETRO_TAPPING
   ```
- Add the following to `crkbd/keymaps/default/keymap.c`:
   ```c
   #define KC_SandA ALT_T(KC_SPC)
   ```
- Replace `KC_RALT` with `KC_SandA` to `_QWERTY` layer in `crkbd/keymaps/default/keymap.c`.
- Compile and write the firmware to crkbd
- Press the `KC_SandA` key over `TAPPING_TERM` and release it.
  - **Expected behavior**: A space is sent (we will see a space).
  - **Actual behavior**: Alt is sent (I saw that a window menu is focused on) rather than a space is sent.

## System Information

 - Keyboard: crkbd
   - Revision (if applicable): default
 - Operating system: Windows 10
 - AVR GCC version: avr-gcc (GCC) 5.4.0

 - ARM GCC version: arm-none-eabi-gcc (15:6.3.1+svn253039-1build1) 6.3.1 20170620

 - QMK Firmware version: 0.8.192

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [x] Other: Nothing

## Additional Context

I confirmed that `SFT_T` and `CTL_T` work as intended.

precondition, 2020-10-10:
Are you still noticing this problem? I couldn't reproduce it

tom-tan, 2020-11-29:
Unfortunately it still remains in 81164c1.

precondition, 2020-11-29:
Are you sure that the space does not simply get "eaten" by the key sequence initialized by alt? When retro-tapping, `KC_LALT` still gets pressed and released before the letter is sent. You should test with a key event listener like https://config.qmk.fm/#/test or `xev` on Linux

kazbeel, 2022-01-10:
I use home row mod keys for Linux (most of the time), Windows (from time to time) and Mac (at work). I do have activated `RETRO_TAPPING_PER_KEY` and `get_retro_tapping` returning `true` for all those mod keys. In both, Linux and Windows, pressing ALT or GUI without a secondary key press results in ALT+S and GUI+D which focuses on the app's menu or activates the System Menu respectively.

This behavior is quite annoying. Is this the intended behavior? Why SFT and CTL do not behave the same way? Should they?

Why the behavior of ALT and GUI differ from CTL and SFT?

precondition, 2022-01-10:
It is common convention in desktop UX to focus the menu and enter "leader key mode" when pressing and releasing alt by itself (i.e. without combining alt with any other key) as I've explained so yes this is the intended behaviour that's coded by the app developers. There is no common convention regarding any special behaviour to tapping control or shift by itself so you rarely find apps that map an action to such a key sequence.

kassi-moon-whaler, 2022-01-13:
I'm experiencing the same issue, it really throws me off when I hold the `ALT_T(KC_SPC)` just a moment too long and the focus is lost from where I'm typing. The functionality I'm looking for is to never send alt. I want to add alt as a modifier to any other keys clicked when I'm holding that key though. 

With your explanation @precondition I understand why the alt modifier works how it does, but is there any work around you can think of to accomplish the desired behavior? 

I've considered using a layer toggle and an alt layer that would be my normal keymap with alt to every key. That would be very difficult to maintain edge cases like Alt+F4 though, since that's on a separate layer. Essentially I'd need an Alt+Function keys layer that is activated from my function layer toggle key when in the alt layer. I'd also need to remember I have that weird pattern when changing other things about my keymap.

kazbeel, 2022-01-13:
Wow, @kassi-moon-whaler! I will give it a try tonight! I'll give feedback then!

kazbeel, 2022-01-25:
I totally forgot to come back here. @kassi-moon-whaler, your solution works like a charm. Thanks so much! You made my day!