qmk/qmk_firmware #9231: a problem with flashing keyboard on macos El Capitan

banihaykal, 2020-05-29:
hello!

i'm very new to flashing keyboards / firmwares etc, and encountered a problem whilst doing my first project. this is the error message i received:

>>> dfu-programmer atmega32u4 erase --force
    dyld: lazy symbol binding failed: Symbol not found: _clock_gettime
      Referenced from: /Applications/QMK Toolbox.app/Contents/Resources/../Frameworks/libusb-1.0.0.dylib (which was built for Mac OS X 10.13)
      Expected in: /usr/lib/libSystem.B.dylib
    
    dyld: Symbol not found: _clock_gettime
      Referenced from: /Applications/QMK Toolbox.app/Contents/Resources/../Frameworks/libusb-1.0.0.dylib (which was built for Mac OS X 10.13)
      Expected in: /usr/lib/libSystem.B.dylib
    
>>> dfu-programmer atmega32u4 flash /Users/admin/Desktop/idobo_idobo_resetv5.hex
    dyld: lazy symbol binding failed: Symbol not found: _clock_gettime
      Referenced from: /Applications/QMK Toolbox.app/Contents/Resources/../Frameworks/libusb-1.0.0.dylib (which was built for Mac OS X 10.13)
      Expected in: /usr/lib/libSystem.B.dylib
    
    dyld: Symbol not found: _clock_gettime
      Referenced from: /Applications/QMK Toolbox.app/Contents/Resources/../Frameworks/libusb-1.0.0.dylib (which was built for Mac OS X 10.13)
      Expected in: /usr/lib/libSystem.B.dylib
    
>>> dfu-programmer atmega32u4 reset
    dyld: lazy symbol binding failed: Symbol not found: _clock_gettime
      Referenced from: /Applications/QMK Toolbox.app/Contents/Resources/../Frameworks/libusb-1.0.0.dylib (which was built for Mac OS X 10.13)
      Expected in: /usr/lib/libSystem.B.dylib
    
    dyld: Symbol not found: _clock_gettime
      Referenced from: /Applications/QMK Toolbox.app/Contents/Resources/../Frameworks/libusb-1.0.0.dylib (which was built for Mac OS X 10.13)
      Expected in: /usr/lib/libSystem.B.dylib

i'm not sure what this means and how to go about troubleshooting this. any help is appreciated!

thank you!

fauxpark, 2020-05-29:
The flashing tools bundled with the Toolbox are taken from the Homebrew bottles, which only supports High Sierra and up: https://docs.brew.sh/Installation

closed 2020-08-28 by fauxpark

nckramer, 2020-12-01:
i got the same error. 1st time i've seen it. i too am still on 10.11.6 (highest this machine will go up to)
i'm familiar with qmk and don't have trouble flashing any of my other devices.

any assistance would be greatly appreciated.

fauxpark, 2020-12-01:
@nckramer I'm afraid you're out of luck, unfortunately. As my link above states, Homebrew now no longer runs on anything older than Mojave, and the bottles we use in the Toolbox are (for now) built for High Sierra and up.

nckramer, 2020-12-01:
eh, i see what it was.... my other ones were caterina, the one that was throwing the issue is dfu... duh...
thanks