qmk/qmk_firmware #9246: [Bug] CannonKeys STM32F072 boards don't build due to missing printf.h

bcat, 2020-05-30:
## Describe the Bug

At QMK head after the latest breaking changes:

`bcat@holden MINGW64 ~/qmk_firmware
$ make clean
QMK Firmware 0.9.0
Deleting .build/ ... done.

bcat@holden MINGW64 ~/qmk_firmware
$ make cannonkeys/an_c:default
QMK Firmware 0.9.0
WARNING: Some git submodules are out of date or modified.
 Please consider running make git-submodule.

Making cannonkeys/an_c with keymap default

arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/cannonkeys/an_c/an_c.c                                                        In file included from tmk_core/common/debug.h:22,
                 from quantum/keymap.h:35,
                 from quantum/quantum.h:30,
                 from keyboards/cannonkeys/an_c/an_c.h:3,
                 from keyboards/cannonkeys/an_c/an_c.c:1:
tmk_core/common/print.h:75:18: fatal error: printf.h: No such file or directory
 #        include "printf.h"  // lib/printf/printf.h
                  ^~~~~~~~~~
compilation terminated.
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:386: .build/obj_cannonkeys_an_c_default/keyboards/cannonkeys/an_c/an_c.o] Error 1
Make finished with errors
make: *** [Makefile:584: cannonkeys/an_c:default] Error 1

bcat@holden MINGW64 ~/qmk_firmware
$ make cannonkeys/instant60:default
QMK Firmware 0.9.0
WARNING: Some git submodules are out of date or modified.
 Please consider running make git-submodule.

Making cannonkeys/instant60 with keymap default

arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/cannonkeys/instant60/instant60.c                                              In file included from tmk_core/common/debug.h:22,
                 from quantum/keymap.h:35,
                 from quantum/quantum.h:30,
                 from keyboards/cannonkeys/instant60/instant60.h:3,
                 from keyboards/cannonkeys/instant60/instant60.c:1:
tmk_core/common/print.h:75:18: fatal error: printf.h: No such file or directory
 #        include "printf.h"  // lib/printf/printf.h
                  ^~~~~~~~~~
compilation terminated.
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:386: .build/obj_cannonkeys_instant60_default/keyboards/cannonkeys/instant60/instant60.o] Error 1
Make finished with errors
make: *** [Makefile:584: cannonkeys/instant60:default] Error 1
`

## System Information

 - Keyboard:  CannonKeys AN-C, Instant60
   - Revision (if applicable):  
 - Operating system: Windows 10
 - AVR GCC version: avr-gcc.exe (GCC) 8.3.0

 - ARM GCC version: arm-none-eabi-gcc.exe (GNU Tools for Arm Embedded Processors 8-2019-q3-update) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]

 - QMK Firmware version: 0.9.0

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

It's been a month about a month since I resynced, so in theory this could have broken any time in that interval, though I suspect the breaking changes first.

bcat, 2020-05-30:
@awkannan FYI

bcat, 2020-05-30:
Nevermind, I totally missed the outdated submodules. :( Confirmed `make git-submodule` fixes this. Sorry for spam!

closed 2020-05-30 by bcat

nicocesar, 2020-07-18:
thanks bcat to take the time to add the last comment... after 30 minutes of chasing my btail.