qmk/qmk_firmware #9248: backlight_driver_common.c: Backlight pin/pins not defined

jericdeleon, 2020-05-31:
# Issue
Intended to compile with `qmk compile -kb crkbd -km default`.
Only thing added was `BACKLIGHT_ENABLE = yes` in `rules.mk`.
Instead of getting a clean compile, error below is encountered:

```
➜ qmk compile -kb crkbd -km default
Ψ Compiling keymap with make crkbd:default

QMK Firmware 0.9.0
Making crkbd/rev1 with keymap default

avr-gcc (GCC) 5.4.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/crkbd/ssd1306.c                                                                [OK]
Compiling: keyboards/crkbd/rev1/matrix.c                                                            [OK]
Compiling: keyboards/crkbd/rev1/split_util.c                                                        [OK]
Compiling: keyboards/crkbd/rev1/split_scomm.c                                                       [OK]
Compiling: keyboards/crkbd/lib/glcdfont.c                                                           [OK]
Compiling: keyboards/crkbd/lib/rgb_state_reader.c                                                   [OK]
Compiling: keyboards/crkbd/lib/layer_state_reader.c                                                 [OK]
Compiling: keyboards/crkbd/lib/logo_reader.c                                                        [OK]
Compiling: keyboards/crkbd/lib/keylogger.c                                                          [OK]
Compiling: keyboards/crkbd/crkbd.c                                                                  [OK]
Compiling: keyboards/crkbd/rev1/rev1.c                                                              [OK]
Compiling: keyboards/crkbd/keymaps/default/keymap.c                                                 [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/debounce/sym_g.c                                                                 [OK]
Compiling: quantum/backlight/backlight.c                                                            [OK]
Compiling: quantum/process_keycode/process_backlight.c                                              [OK]
Compiling: quantum/backlight/backlight_driver_common.c                                             quantum/backlight/backlight_driver_common.c:6:6: error: #error "Backlight pin/pins not defined. Please configure."
 #    error "Backlight pin/pins not defined. Please configure."
      ^
quantum/backlight/backlight_driver_common.c:22:36: error: ‘BACKLIGHT_PIN’ undeclared here (not in a function)
 static const pin_t backlight_pin = BACKLIGHT_PIN;
                                    ^
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:386: .build/obj_crkbd_rev1_default/quantum/backlight/backlight_driver_common.o] Error 1
make: *** [Makefile:584: crkbd:default] Error 1
Make finished with errors
```

## System Information
 - Keyboard:  `crkbd`
   - Revision (if applicable):  `rev1`
 - Operating system: 
```
➜ lsb_release -a
No LSB modules are available.
Distributor ID:	Pop
Description:	Pop!_OS 20.04 LTS
Release:	20.04
Codename:	focal
```
 - QMK Firmware version: `0.9.0`
```
➜ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ Detected Linux.
⚠ Missing udev rules for 'dfu' boards. You'll need to use `sudo` in order to flash them.
⚠ Missing udev rules for 'input_club' boards. You'll need to use `sudo` in order to flash them.
⚠ Missing udev rules for 'stm32' boards. You'll need to use `sudo` in order to flash them.
⚠ Missing udev rules for 'bootloadhid' boards. You'll need to use `sudo` in order to flash them.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Missing udev rules for 'caterina' boards. You'll need to use `sudo` in order to flash them.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ Problems detected, please fix these problems before proceeding.
```

zvecr, 2020-05-31:
Backlight is not supported on that board so that would be an expected error. It either runs rgb light or rgb matrix.

jericdeleon, 2020-05-31:
I see, so it seems was using an incorrect `rules.mk` in the first place; I had `BACKLIGHT_ENABLE = yes` ever since 2 years ago, and it did not raise an error when `make crkbd:default` was the process, and not `qmk compile`

Thanks!

closed 2020-05-31 by jericdeleon