qmk/qmk_firmware #9279: [Bug] Trouble getting udev rules updated

Selovert, 2020-06-03:
## Describe the Bug
When I'm running 'qmk setup' I ran into an exception on udev rules. I've followed the instructions and added the rules to /etc/udev/rules.d/ but the error is persisting. 

```
tseloverstephan@tstephan-lvm:~$ qmk setup
Ψ Found qmk_firmware at /home/tseloverstephan/qmk_firmware.

*** Would you like to set /home/tseloverstephan/qmk_firmware as your QMK home? [y/n] y

Ψ Wrote configuration to /home/tseloverstephan/.config/qmk/qmk.ini
Ψ QMK Doctor is checking your environment.
Ψ Detected Linux.
⚠ Found old, deprecated udev rules for 'dfu' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
⚠ Found old, deprecated udev rules for 'input_club' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
⚠ Found old, deprecated udev rules for 'stm32' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
⚠ Found old, deprecated udev rules for 'bootloadhid' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
⚠ Detected ModemManager without the necessary udev rules. Please either disable it or set the appropriate udev rules if you are using a Pro Micro.
⚠ Found old, deprecated udev rules for 'caterina' boards. The new rules on https://docs.qmk.fm/#/faq_build?id=linux-udev-rules offer better security with the same functionality.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 4.9.3
Ψ Found avr-gcc version 4.9.2
Ψ Found avrdude version 6.2
Ψ Found dfu-util version 0.8
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ Problems detected, please fix these problems before proceeding.
```

Here's the contents of /etc/udev/rules.d/:
```
tseloverstephan@tstephan-lvm:~$ ls -al /etc/udev/rules.d/
total 96
drwxr-xr-x 2 root root  4096 Jun  2 17:06 .
drwxr-xr-x 4 root root  4096 Jun  2 14:17 ..
-rw-r--r-- 1 root root   314 Jun  2 16:23 50-atmel-dfu.rules
-rw-r--r-- 1 root root   121 Jun  2 16:24 52-tmk-keyboard.rules
-rw-r--r-- 1 root root    90 Jun  2 16:24 54-input-club-keyboard.rules
-rw-r--r-- 1 root root   159 Jun  2 16:24 55-caterina.rules
-rw-r--r-- 1 root root   195 Jun  2 16:25 56-dfu-util.rules
-rw-r--r-- 1 root root    98 Jun  2 16:34 57-bootloadhid.rules
-rw-r--r-- 1 root root   134 Sep 23  2016 60-vboxadd.rules
-rw-r--r-- 1 root root 58549 Aug  7  2019 70-snap.core.rules
```

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system: Ubuntu 16.04.6 LTS
 - AVR GCC version:  avr-gcc (GCC) 4.9.2
 - ARM GCC version: arm-none-eabi-gcc (15:4.9.3+svn231177-1) 4.9.3 20150529 (prerelease)
 - QMK Firmware version: 0.9.5
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

zvecr, 2020-06-03:
The updated udev checks got merged before #8750 which updates the recommendations.

For end users we now have a situation where the docs dont help. Short term you can try to use the suggestions within #8750, or if you can flash ignore for now

ginxxxxx, 2020-06-05:
i trusted qmk's directions and blindly created the files / ran the commands.

ginxxxxx, 2020-06-06:
both the qmk docs and #8750 did not work for me as flashing waited forever even when my device was resetting. sudo didnt work or any combination of commands.

I just removed all of it and used sudo instead. you still have to stop/disable/purge the modemmanager service or it will fail at writing

Erovia, 2020-06-07:
#8750 just got merged.

@ginxxxxx Can you do a few things to see where the problem lies?
After setting the rules from the (now updated) docs, check the permission for the device for the bootloader.

* If it's `dfu`, check the bus and devnum values with `lsusb`. You should see something like this:
```
Bus 002 Device 012: ID 03eb:2ff4 Atmel Corp. atmega32u4 DFU bootloader
```
In this case, bus is 002 and devnum is 012, so check the permissions:
```
getfacl /dev/bus/usb/002/012
getfacl: Removing leading '/' from absolute path names
# file: dev/bus/usb/002/012
# owner: root
# group: root
user::rw-
user:erovia:rw-
group::rw-
mask::rw-
other::r--
```

* If it's `caterina` (in case of a Pro Micro):
```
getfacl /dev/ttyACM0
getfacl: Removing leading '/' from absolute path names
# file: dev/ttyACM0
# owner: root
# group: dialout
user::rw-
user:erovia:rw-
group::rw-
other::---
```

In any case, your user should have read-write access via ACL.

closed 2022-08-28 by zvecr