qmk/qmk_firmware #9293: No link to full keycode list and other references.

mcp292, 2020-06-04:
When following [this section](https://docs.qmk.fm/#/newbs_building_firmware?id=customize-the-layout-to-your-liking) in the tutorial, and then going to [basic keycodes](https://docs.qmk.fm/#/keycodes_basic?), there is no link or reference to the full list, nor links to other resources for more advanced customization. This would be nice because it makes the reader aware that there's more capability if they need it, and gives them a way to get back to it when they are ready.

nblyumberg, 2020-06-05:
There **is** a reference to more advanced features in that section:
`explore the sidebar to the left under “Using QMK” to see the full list.`

I initially started to add the advanced features as a list, but then I realized that this may cause more problems down the road because I would be, for the most part, recreating the navigation bar section. This would be just another spot to maintain when documentation is revised.

mcp292, 2020-06-05:
Okay, that should be fine. I didn't see it when returning for reference. I focused on the links. Would you consider bolding it?

closed 2020-10-04 by stale[bot]