qmk/qmk_firmware #9301: Toolchain setup -- unrecognized options

cvanelteren, 2020-06-04:
I am building on arch and followed setup using the readme

`avr-gcc --version // 8.3.0`
`gcc --version //10.1.0 `

however I am getting unrecognized options when building for teensy firmware. In particular 
`-march=nocona` (gcc) and `-mtune=haswell` (avr-gcc) is not recognized. I am not too familiar with avr-gcc but does it need to have a matching gcc version? Why is the option not recognized. I have used qmk in the past and this is the first time I am having this issue. 

[image]

cvanelteren, 2020-06-04:
In addition the output from `sh util/qmk_install.sh` states all dependencies are up-to-date.

fauxpark, 2020-06-04:
Neither of those options are specified in QMK. It sounds like something else is interfering - we don't use regular gcc at all, only avr-gcc and arm-none-eabi-gcc.

cvanelteren, 2020-06-04:
Fixed it. I installed it accidentally using conda vritual env. Using system python it compiles. Can be closed.

closed 2020-06-04 by cvanelteren

jfass, 2021-06-12:
@cvanelteren I'm having this problem in conda venv's or not ... what did you install accidentally, qmk? I can uninstall qmk, blow away the firmware repo, remove my (Ubuntu) avrdude and gcc-avr packages, then reinstall, it asks to install dependencies or I run utils/qmk_install.sh and I still get this error. Can you share your wisdom?

cvanelteren, 2021-06-13:
@jfass  It's been some time since I last compiled with qmk. Unfortuately, I can't recall the steps that I took. From my previous comment I can deduce that I may have installed it while being in my conda env (which is on by default). I am guessing past me reinstalled everything while out of the env.