qmk/qmk_firmware #9309: [Bug] Planck EZ linking error: “collect2: error: ld returned 1 exit status”

Lilalas, 2020-06-05:
I’m trying to compile the default keymap of the Planck EZ from the QMK repo.

## Describe the Bug

This is the output of the build command:

```
$ qmk compile -kb planck/ez -km default
Ψ Compiling keymap with make planck/ez:default

QMK Firmware 0.8.144
Making planck/ez with keymap default

arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/planck_ez_default.elf                                                               [ERRORS]
 | 
 | /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-abort.o): in function `abort':
 | /build/newlib-CVVEyx/newlib-3.3.0/build_nano/arm-none-eabi/thumb/v7e-m/nofp/newlib/libc/stdlib/../../../../../../../../newlib/libc/stdlib/abort.c:59: undefined reference to `_exit'
 | /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-signalr.o): in function `_kill_r':
 | /build/newlib-CVVEyx/newlib-3.3.0/build_nano/arm-none-eabi/thumb/v7e-m/nofp/newlib/libc/reent/../../../../../../../../newlib/libc/reent/signalr.c:53: undefined reference to `_kill'
 | /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg_nano.a(lib_a-signalr.o): in function `_getpid_r':
 | /build/newlib-CVVEyx/newlib-3.3.0/build_nano/arm-none-eabi/thumb/v7e-m/nofp/newlib/libc/reent/../../../../../../../../newlib/libc/reent/signalr.c:83: undefined reference to `_getpid'
 | collect2: error: ld returned 1 exit status
 | 
make[1]: *** [tmk_core/rules.mk:306: .build/planck_ez_default.elf] Error 1
Make finished with errors
make: *** [Makefile:579: planck/ez:default] Error 1
```

When I tried to compile the first time, the output was a little longer (it successfully compiled a bunch of files beforehand), but the error message was the same.

## System Information

I’ve setup my build environment using `pip install`. Package is up to date (version 0.0.35).

 - Operating system: Ubuntu 20.04.
 - AVR GCC version: 5.4.0.
 - ARM GCC version: see above.
 - QMK Firmware version: see above.

## Additional Context

I also tried to compile a few other default keymaps, all of which worked fine. So the issue is most likely related to this particular keyboard/keymap.

Lilalas, 2020-06-05:
@fauxpark That’s bad, thank you for pointing that out. As I mentioned above, I’ve setup my build environment using `pip`, not `git`. I will have to look into building QMK from `git`. Unfortunately, this is not documented on [docs.qmk.fm](https://docs.qmk.fm/#/newbs_getting_started) though, is it?

fauxpark, 2020-06-05:
You don't have to, just `cd` to your qmk_firmware directory and run those two commands.

Lilalas, 2020-06-05:
You were right. After updating it compiles without problem!

```
$ qmk compile -kb planck/ez -km default
Ψ Compiling keymap with make planck/ez:default

QMK Firmware 0.9.7
Making planck/ez with keymap default

arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text	   data	    bss	    dec	    hex	filename
      0	  62268	      0	  62268	   f33c	.build/planck_ez_default.hex

Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/planck_ez_default.elf                                                               [OK]
Creating binary load file for flashing: .build/planck_ez_default.bin                                [OK]
Creating load file for flashing: .build/planck_ez_default.hex                                       [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  62268	      0	  62268	   f33c	.build/planck_ez_default.hex

Copying planck_ez_default.bin to qmk_firmware folder                                                [OK]
(Firmware size check does not yet support cortex-m4 microprocessors; skipping.)
```

Lilalas, 2020-06-05:
So the issue came down to me not understanding how to keep my installation up to date. Let me mention again that I think this is poorly (not at all) explained in the documentation. I will create a separate issue over there.

closed 2020-06-05 by Lilalas

fauxpark, 2020-06-05:
It's under the "Git Best Practices" section: https://docs.qmk.fm/#/newbs_git_using_your_master_branch

timvisee, 2020-11-09:
I'm getting the same (or at least a similar) error. I'm using QMK 0.10.50.

```bash
$ ~/.local/bin/qmk compile
Ψ Compiling keymap with gmake ergodox_infinity:timvisee

QMK Firmware 0.10.50
Making ergodox_infinity with keymap timvisee

arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/ergodox_infinity/keymaps/timvisee/keymap.c                                     [OK]
Compiling: quantum/process_keycode/process_terminal.c                                               [OK]
Compiling: tmk_core/common/command.c                                                                [OK]
Linking: .build/ergodox_infinity_timvisee.elf                                                       [ERRORS]
 |
 | /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-abort.o): in function `abort':
 | /build/newlib-CVVEyx/newlib-3.3.0/build/arm-none-eabi/thumb/v7e-m/nofp/newlib/libc/stdlib/../../../../../../../../newlib/libc/stdlib/abort.c:59: undefined reference to `_exit'
 | /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-signalr.o): in function `_kill_r':
 | /build/newlib-CVVEyx/newlib-3.3.0/build/arm-none-eabi/thumb/v7e-m/nofp/newlib/libc/reent/../../../../../../../../newlib/libc/reent/signalr.c:53: undefined reference to `_kill'
 | /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/bin/ld: /usr/lib/gcc/arm-none-eabi/9.2.1/../../../arm-none-eabi/lib/thumb/v7e-m/nofp/libg.a(lib_a-signalr.o): in function `_getpid_r':
 | /build/newlib-CVVEyx/newlib-3.3.0/build/arm-none-eabi/thumb/v7e-m/nofp/newlib/libc/reent/../../../../../../../../newlib/libc/reent/signalr.c:83: undefined reference to `_getpid'
 | collect2: error: ld returned 1 exit status
 |
gmake[1]: *** [tmk_core/rules.mk:306: .build/ergodox_infinity_timvisee.elf] Error 1
Make finished with errors
gmake: *** [Makefile:584: ergodox_infinity:timvisee] Error 1
```

I did pull in the newest changes and did run `make git-submodules`. I'm building on Ubuntu 20.10 for an 'Ergodox Infinity'. All my system packages are up-to-date.