qmk/qmk_firmware #9317: ATMEGA32A ISSUE

therealstoia, 2020-06-06:
Hi guys. I am triyng to flash a gh60 minila (atmega32a) that has software issues.

I am using a pro micro as an isp and it seems to work fine, but when i use qmk to flash the firmware this message is shown to me:

"avr part atmega32a not found."

even if the flash is shown as complete

idk how to solve that...

therealstoia, 2020-06-06:
I managed to flash the firmware by coding in cmd, but after I flashed the keymap hex file with bootloader, keyboard didn't work properly.

It doesn't type at all, but program recognised the lcrtl pression when I flash😰

fauxpark, 2020-06-06:
Could you post a screenshot of this error?

I don't know what a "GH60 minila" is, but maybe you're referring to the [YD60](https://www.aliexpress.com/item/32799437588.html)? The sellers on Aliexpress aren't the best with naming.

therealstoia, 2020-06-06:
Hi bro. I got no screenshots, bc there is no advise of errors. 

I flashed the file, than when I type the kB doesn't work.
I got no error shown by windows, but if I try to download the keymap that I flashed, bootmapper doesn't recognize the keyboard as plugged.

therealstoia, 2020-06-06:
Here you are the link of the article
[https://a.aliexpress.com/_dVsZ9j4]()

fauxpark, 2020-06-06:
That is a YMDK Bface: https://github.com/qmk/qmk_firmware/tree/master/keyboards/ymdk/bface/

therealstoia, 2020-06-07:
YEAH it is. Now Idk how to go on. I flashed the bootload software using the cmd, i flashed the keymap hex file using bootloader holding left ctrl while plugging. If i did not hold it, then the keyboard is not recognized.

You gave me some code to flash, but idk how to use it.. may i download visual studio code? How it works? thanks

therealstoia, 2020-06-07:
actually i am pretty sure i flashed it correctly, but it still doesn't work

fauxpark, 2020-06-07:
It would be really helpful if you could provide some screenshots. I'm not sure what you mean by flashing the "bootload software".

The link I posted is simply showing that the keyboard name in QMK is `ymdk/bface`, so that is the one you want to select in the Configurator.

therealstoia, 2020-06-07:
Ok. I'll link a video.

Resuming: I use this guide [](https://beta.docs.qmk.fm/using-qmk/guides/keyboard-building/isp_flashing_guide) to flash the bootloader file with an isp in cmd.

I used the software the seller recommended to use since bootloadhid.exe does't run on my pc (error 33) and i couldn't fix it.[](https://drive.google.com/file/d/1tK2jK3pWR7nLH4nmj429yniL5HXjYAn1/view?usp=sharing)

therealstoia, 2020-06-07:
i was isp flashing because windows didn't recognize the pcb. I wanted to use qmk, but the chip is an Atmega32a and qmk toolbox didn't work

fauxpark, 2020-06-07:
You need to provide more information, "qmk toolbox didn't work" is not all that helpful on its own. A screenshot of the Toolbox "not working" would be nice.

therealstoia, 2020-06-07:
I was wrong bacause i used the wrong way to flash it i guess.

I have been trying using msys2, and now i have to flash te file into the keyboard. This is the issue.[
[image]
](url)

spidey3, 2020-06-07:
You should be able to flash by doing `qmk flash -kb ymdk/bface -km default`...

therealstoia, 2020-06-08:
I'll try when I go back home from work, thanks.

However I have a doubt: how can I reset the kB? jumping jb holes is ok? Bc I tried yesterday before sleeping, but no message of "device found" appeared

therealstoia, 2020-06-08:
IT DOES WORK! Thank you both guys.

fauxpark, 2020-07-06:
@therealstoia can this be closed?

therealstoia, 2020-07-06:
> @therealstoia can this be closed?

Yeah

closed 2020-07-06 by therealstoia