qmk/qmk_firmware #9320: [Bug] PS2_MOUSE_ROTATE not affecting mouse movements

GSchiaffonati, 2020-06-06:
Defining PS2_MOUSE_ROTATE 90 or 180 in my config.h has no effect on the orientation of the mouse (trackpoint). This is the config.h: qmk/keyboards/<my_trackpoint>/config.h
Using AVR USART for the PS/2 connection it is always operating in the normal orientation.

## Describe the Bug

config.h
```c
#define PS2_MOUSE_ROTATE 90
```

Has no effect on the direction of the mouse.
 

## System Information

 - Keyboard:  teensy 2.0 with only a PS/2 trackpoint using USART
   - Revision (if applicable):  N/A
 - Operating system: Windows 10, Ubuntu 20.04
 - AVR GCC version: 5.4.0
 - ARM GCC version: 9.2.1
 - QMK Firmware version: 0.7.152
 - Any keyboard related software installed? 
no other keyboard software.

## Additional Context

The documentation:
https://docs.qmk.fm/#/feature_ps2_mouse?id=ps2-mouse-features-idps2-mouse-features
suggests I have defined this in the correct file along with the other definitions for my PS2 mouse connections i.e pin and port definitions.

Not sure if this is a QMK, TMK or ChibiOS problem. Or if it's something I've done wrong...

fauxpark, 2020-06-06:
Your repo is outdated. `ps2_mouse.c` at tag 0.7.152 (Feb 14) makes no mention of `PS2_MOUSE_ROTATE`:
https://github.com/qmk/qmk_firmware/blob/0.7.152/tmk_core/protocol/ps2_mouse.c
as it was added in #8650 from April 2.

GSchiaffonati, 2020-06-06:
Thanks, that was dumb of me...

I didn't even realise I started this little project that long ago!

closed 2020-06-06 by GSchiaffonati