qmk/qmk_firmware #9323: Issue with BM43A rules.mk file[Bug]

DJDiarrhea, 2020-06-06:
## Describe the Bug

Backlight and underglow on the bm43a doesn't work with stock firmware

BACKLIGHT_ENABLE = no 
RGBLIGHT_ENABLE = no  
Should be set to yes as the keyboard has underglow aswell as backlighting.

RmigD, 2020-12-03:
Any other way of getting the 6 back LEDs to light up? Read something about having to use libusb, but I have no idea how that would work...

DJDiarrhea, 2020-12-03:
Not that I know of. Compiling the firmware yourself is pretty straightforward though. There is a guide on the qmk GitHub how to do it. From: RmigDSent: Thursday, 3 December 2020 01:15To: qmk/qmk_firmwareCc: DJDiarrhea; AuthorSubject: Re: [qmk/qmk_firmware] Issue with BM43A rules.mk file[Bug] (#9323) Any other way of getting the 6 back LEDs to light up? Read something about having to use libusb, but I have no idea how that would work...

RmigD, 2020-12-03:
Yeah, I deleted my comment because I realized it's an easy fix. Thanks.

closed 2022-08-28 by zvecr