qmk/qmk_firmware #9347: [Bug] Boardwalk user keymap doesn't compile

pyonium, 2020-06-09:
In PR #9341 a keymap for Boardwalk in userspace (users/yet-another-developer) failed to compile. Should this one be removed?

pyonium, 2020-06-09:
I'm seeing it happen on other keymaps too. Might be a decent idea to go through them all and flag which ones don't work

drashna, 2020-07-03:
That's fine.  It's not something we are worried about. 

Specifically, as long as the default keymaps compile, then it's okay. 
And in this case, it was a community layout that isn't compiling.  And it's a LOT harder to make sure that these always compile, just due to their nature.  Specifically that it is a single keymap for multiple, different boards.  

Also, if a user has changes that haven't been pushed upstream (extremely common), then we are forcing users to handle merge conflicts or use the xkcd method of updating. 

So, errors on user keymaps are generally ignored, and are considered okay. 
Eg, as long as `default` `default_*` and `via` compile, it's not important.  

Absolutely worst case, you can tag the users and let them know.  In this case @yet-another-developer, it looks like some of the rgb stuff is erroring out on the boardwalk board.  This is because the rgb stuff in the rules.mk was removed, but the rest wasn't. 

Also, it may help to run `make all:yet-another-developer`, as this will compile your keymap for every board that supports it.

closed 2022-08-28 by zvecr