qmk/qmk_firmware #9350: [Bug] Murcielago slave not working anymore

elagil, 2020-06-09:
## Describe the Bug

Murcielago is a split keyboard that works with UART and uses EEPROM side detection. Therefore, its config.h includes:

`
#define EE_HANDS
#define SPLIT_USB_DETECT
#define SOFT_SERIAL_PIN E6
`

Previously, this worked fine. With a recent QMK update, the slave does not work anymore.

Before: QMK configurator default firmware worked
Now: Slave is broken

I reflashed the split EEPROM image to both sides, just to be sure, but it didn't help.

## System Information

 - Keyboard:  Murcielago
   - Revision (if applicable): Rev1  
 - Operating system: Windows 
 - AVR GCC version: 5.4.0

 - QMK Firmware version: 0.9.7
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [x] None

elagil, 2020-06-09:
I found that functionality breaks from https://github.com/qmk/qmk_firmware/releases/tag/0.8.175 onwards, where there are some changes to https://github.com/qmk/qmk_firmware/compare/0.8.174...0.8.175#diff-57e6bd7db0b3e56c607994b241d7b945

semnil, 2020-06-10:
The following changes will work. 🤔 

https://github.com/semnil/qmk_firmware/commit/ccc1b6b44c05c4602e5cfa90fe2b1b6b768dff10

It hasn't worked since this commit. 😭 

https://github.com/qmk/qmk_firmware/commit/1816ad01d0efcc987b5aa7a833331b0f9d903d4f

@zvecr What do you think about this?