qmk/qmk_firmware #9357: [Bug] Building fails if lang environment variable if not en_US.UTF-8

krautcat, 2020-06-10:
## Describe the Bug

I don't know whether it is documented or not, whether it is bug or not but. QMK building fails if I have `LANG` environment variable not set to `en_US.UTF-8`.

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system: Ubuntu 20.04
 - AVR GCC version: 5.4.0

 - ARM GCC version: 

 - QMK Firmware version: 0.9.7

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I have `LANG` envvar setted to `ru_RU.UTF-8`. When I try to build firmware for Iris with any keymap (even from upstream), build fails on linkage stage saying that `.elf` file is empty. Yes, it is really empty, and linkage than fails. Object files are Vgenerated. TBH, I didn't dig into build process of QMK, so I am just assuming is error on linkage stage. Here is log with `VERBOSE=True`. It contains messages in Russian but they are not essential.

```
Linking: .build/keebio_iris_rev2_krautcat.elf                                                       [ERRORS]
 | 
 | collect2: фатальная ошибка: ld завершилась по сигналу 6 [Аварийный останов], core dumped
 | компиляция прервана.
 | *** buffer overflow detected ***: terminated
 | 
make[1]: *** [tmk_core/rules.mk:306: .build/keebio_iris_rev2_krautcat.elf] Ошибка 1
if [ -f .//.build/error_occurred ]; then printf "\033[31;01mMake finished with errors\n\033[0m" & exit 1; fi;
```

When I try to build with `LANG` envvar set to `en_US.UTF-8`, firmware is building, no linkage errors here.

If you don't have any time for investigating the possible bug or it is expected behaviour, please let me know.

rmwphd, 2020-06-11:
Does it fail iff LANG is not set to `en_US.UTF-8` or does it fail iff LANG is set to `ru_RU.UTF-8`? Did you try with other encodings?

krautcat, 2020-06-11:
I haven't tried any other encodings, or, to be more correct, any other values of this environment variables. Tbh, I set it to the `en_US.UTF-8` just to get more meaningful message when linkage fails but apparently this also fixed whole build process.

I will try different locales. Then I will try to unset only `LANG` envvar without touching the `LC_*` envvars (they override `LANG` values for different aspects of localization). Finally, I will try to build with all variables unset.

teuchezh, 2021-10-12:
Same issue, command `export LANG=en_US.UTF-8` before compile fix the problem.

closed 2022-08-29 by zvecr