qmk/qmk_firmware #9379: [Bug] Layer macros with no effect and layer function layer_invert doesnt work as expected

Nesuma, 2020-06-11:
## Describe the Bug

Im trying to use a second layer on my keyboard that I can toggle on and off. TG (and all other layer macros) result in "error: statement with no effect" on compilation. Without treating warning as errors it becomes a warning but still doesnt work. If I use layer_invert I can switch between my layers but its only working while pressing the button despite being described as toggling. 

## System Information

 - Keyboard:  Handwired
 - Operating system: Linux archdesktop 5.6.15-arch1-1
 - AVR GCC version: 

 - ARM GCC version: 10.1.0

 - QMK Firmware version: 0.9.10

## Additional Context

rmwphd, 2020-06-11:
Hi! I notice you don't have the code you're talking about uploaded to a forked repo. If you push your code to a fork, we can see what you're talking about and help you with your error. Also, this kind of question will possibly see more and quicker response if you ask on the Discord.

drashna, 2020-07-12:
Also, you're using gcc version 10, which may be part of the problem.

closed 2022-08-28 by zvecr