qmk/qmk_firmware #9456: how can I specify a different MIDI channel for identical MIDI notes on a layer?

ralfkaa, 2020-06-17:
**Background**: I'm implementing MIDI on a hand-wired keyboard. Several notes repeat by design to make for more convenient fingering (I'm using isomorphic note layouts). But if the repeated notes send the same MIDI note keycodes on the same MIDI channel (i.e., if there are notes on the keyboard which are absolutely identical), I run into trouble depending on what DAW and/or OS I'm using.

**Question**: Can I configure a layout (in the keymap, I assume) so that each switch sends not only a MIDI note, but also a MIDI channel specification?

An example using quantum keycodes: on a layer I need to have, say, the note MI_G_2 playing on channel MI_CH1 on one switch, and the **same** MI_G_2 playing on channel MI_CH2 on another switch (on the same layer).

Is that possible, and if so, is there any written documentation on this or example that could show me how to implement this?

Thank you in advance for any help!

AlexOConnorHub, 2020-06-20:
I've looked into QMK's MIDI abilities a bit for my keyboard, and sadly, I've not seen this as an option. I may have missed seeing this in my searching though.

ralfkaa, 2020-06-20:
thanks, Alex. i realize MIDI is still experimental in QMK, but i was hoping someone might have had the same need and found a solution.
i'm going to experiment with macros, if i find a workaround to achieve this i will share it here.

AlexOConnorHub, 2020-06-20:
If you check out my fork, I made Keyboard handwired/ortho5x12. I made a custom midi keyboard layer where I store the octave, and make a 'custom' key that sends its notes at the octave specified. I didn't know that sending the same note at the same octave caused an error, so I need to make some keys send octave +1. You can check out out and see if anything there could help you out.

ralfkaa, 2020-06-20:
it doesn't cause an error, per se, but different DAWs treat a repeated note differently across platforms (i'm using Linux and Mac mainly). on a Mac running GarageBand the notes will both play simultaneously, but it activates a sustain effect that stays until you unplug the keyboard. on LMMS on Linux the second not simply won't play.
what is the name of your keyboard?

AlexOConnorHub, 2020-06-20:
https://github.com/AlexOConnorHub/qmk_firmware is my fork. (in the master branch) my files are in keyboards/handwired/ortho5x12

ralfkaa, 2020-06-20:
that's quite interesting.
you didn't use the quantum keycodes at all, you just defined your MIDI notes in the funcs.c file, then did an "include" in your keymap, is that correct?
i'm very much a dabbler when it comes to coding, so it's a bit difficult to understand how that works, but i presume you could have assigned different MIDI channels in your funcs.c file, couldn't you?
do you have speakers on your keyboard, or do you just play through a DAW (on or offline)?

AlexOConnorHub, 2020-06-20:
You understand that perfectly! I don't use quantum because of the constant channel and velocity. I made channel +/- and velocity +/- buttons. I checked the quantum buttons, and they call a midi send function. So I just call that function, but use what ever velocity /channel is selected currently. I don't have a keyboard yet, waiting on parts, so I keep adding to my keymap, hehe. I don't plan on having a speaker on my board, I'll just use a DAW.

ralfkaa, 2020-06-20:
would you mind my experimenting with a similar solution (provided i can wrap my head around your code)?
for the sake of clarification, i'm implementing an isomorphic ([Wicki-Hayden](https://en.wikipedia.org/wiki/Wicki–Hayden_note_layout)) layout.

AlexOConnorHub, 2020-06-21:
Enjoy! I put it on git for that reason.

ralfkaa, 2020-06-21:
thanks! i will let you know how it goes and share the code (if it works...).

AlexOConnorHub, 2020-06-21:
Hey, I just split up that one file func.c into 3, because the one file was getting long, and I was really making 3 different new functions. now custom_midi.c has what I was doing with midi.

AlexOConnorHub, 2020-06-21:
*My keyboard is now in a branch called "custom-keyboard" on my fork.

ralfkaa, 2020-06-26:
hi Alex, your solution doesn't seem to work for me (perhaps because i understand it conceptually, but not in terms of actually developing the code). i need to set different channels for repeated notes by default.
i've experimented with ways of including two Quantum MIDI keycodes for a single key, and that didn't work.
now i'm looking at defining macros, which seems like a really convoluted way of going about this, but since i'm not a coder, that seems to be the only option open to me right now :-/

ralfkaa, 2020-06-26:
> *My keyboard is now in a branch called "custom-keyboard" on my fork.

hi Alex, i have a technical question for you: there's a way to use a single keystroke to send a KC with a modifier, as in LCTL(kc). 

could i just define LCTL(noteX) as "noteX on MIDI channel 2"? something like "if midi is on, define LCTL(noteX) as noteX played on midi channel 2".

any thoughts on this?

AlexOConnorHub, 2020-06-29:
Sorry I just got around to thinking about this...

So to respond to your first question, with my implementation, repeated notes are not sent on different channels. I just thought my project might provide a base for you to modify. While there might be a more efficient way to do this, one idea I can think of is making a boolean for each note, and if it is true, you could send on channel 2, if not, send on channel 1. I think you understand what my code does rather well, but I may have misrepresented what I made it do.

To reply to your second question, you can check if the left control modifier key is being pressed with `(get_mods() & (MOD_BIT(KC_LCTL))) != 0`. You then just need to add this as an if statement to your code.

ralfkaa, 2020-07-09:
Alex, thank you for your comments, and sorry for the delayed response!
I don't think a boolean for each note is a solution, because we're talking about 128 notes on 16 channels (to accommodate the full range of midi). If you then want to add, say, the possibility of quarter-tone notes, you have another 128 notes (in quarter-tone intervals) across the same 16 channels.
If I understand your last paragraph, in "common language" it would be: "if MIDI is active, define LCTL as MIDI channel 2". That would work for a few channels, but I would soon run out of modifier combinations.
So I really have to create a MIDI-specific set of commands with which I can define any note as "MIDIchannel(MIDInote)" in a keymap file. I suppose that means I'd have to fiddle with the process_midi files, which (not being a coder) puts me at risk of breaking things :-/
I will take a stab at it, and let you know how/if it creates something useful.
Thanks for engaging with me on this challenge!

3araht, 2020-08-29:
> on a Mac running GarageBand the notes will both play simultaneously, but it activates a sustain effect that stays until you unplug the keyboard.

It's a bit off from the main topic here but this happens on my setup, too.
I referred [here](https://github.com/qmk/qmk_firmware/blob/9b232a7f885e320061aae994713acd31297eac5c/keyboards/sentraq/s60_x/keymaps/bluebear/keymap.c#L664) to assign a chord to each keys. It seems to happen when multiple chords are played and when the chords shares the same note(s). I'm looking for a solution to avoid it.
When you are using GarageBand, the symptom can be fixed by clicking "Revert" button on the bottom left corner on the GarageBand window. But want to avoid it from the first place...

ralfkaa, 2020-09-05:
> > on a Mac running GarageBand the notes will both play simultaneously, but it activates a sustain effect that stays until you unplug the keyboard.
> 
> It's a bit off from the main topic here but this happens on my setup, too.
> I referred [here](https://github.com/qmk/qmk_firmware/blob/9b232a7f885e320061aae994713acd31297eac5c/keyboards/sentraq/s60_x/keymaps/bluebear/keymap.c#L664) to assign a chord to each key. It seems to happen when multiple chords are played and when the chords share the same note(s). I'm looking for a solution to avoid it.
> When you are using GarageBand, the symptom can be fixed by clicking "Revert" button on the bottom left corner on the GarageBand window. But want to avoid it from the first place...
> 
> I raised this topic [here](https://github.com/qmk/qmk_firmware/issues/10199#issue-688525281).

the solution is to have the possibility to assign identical notes to different MIDI channels, but qmk doesn't support that functionality (yet). it only supports keyboard-wide channel up/down functions.
that said, if we bypass the higher-level definitions in quantum keycodes and address the underlying midi function, we should be able to add a table for repeat notes across different channels.
if i find a way to do that, i will post it here.

3araht, 2020-09-18:
It seems I found a solution.
Can you take a look at this?
https://github.com/qmk/qmk_firmware/issues/10199#issuecomment-694722815

ralfkaa, 2020-09-18:
> It seems I found a solution.
> Can you take a look at this?
> [#10199 (comment)](https://github.com/qmk/qmk_firmware/issues/10199#issuecomment-694722815)

hi @3araht 3araht, i don't understand what your code does. can you explain the logic of what you did in plain language?

3araht, 2020-09-18:
To be honest, I'm not 100% sure why this modification fixed the issue. I should say it just happened.
I guess something irregular happens when the same key is pressed before the first one being released.
Usually, normal keyboards don't have duplicate keys. The number keys on the upper row and num-pads are individual keys.
Probably handling duplicate keys are not well-developed because of such reason.

What I intended to do was to suppress the second or further midi_send_noteon() if the note is already turned on.
Does this make sense?

ralfkaa, 2020-09-18:
"suppress the second or further midi_send_noteon" makes sense, yes. but that wouldn't even be an issue if we found a way to send the same note on different MIDI channels. that's what i'm still shooting for ;-)

ralfkaa, 2020-11-10:
ok, i'm stuck. 
hoping someone like @jackhumbert might indicate whether what i am trying to do is even possible with @qmk. 
or whether he can point me to someone who might be able to help with this MIDI requirement.

brandenbyers, 2020-11-14:
I’m new to the concept of isomorphic midi keyboards but did a quick search and have a basic understanding now. My main question is: what is the benefit of having the same note on multiple channels playing at the same time as opposed to suppressing a duplicate note?

If multiple channels for the same midi key code is a requirement, does every key/note need to play on channel 1 unless it is a duplicate at which point it must play on the next unused channel for that note? Or could different switches that duplicate the same notes be hard coded to separate channels and always play on their respective channels?

How many repeats of the same notes are there on an isomorphic keyboard?

ralfkaa, 2020-11-14:
hi @brandenbyers , thanks for pitching in! and thanks for all the previous refinements you've contributed to the MIDI component!

the benefit is what is called "immutable fingering", meaning: you can move your hand around on a key array when transposing a piece (melody or chord) without having to find a new fingering because the piece doesn't "fit" within the keys you have left.

what you propose in the second paragraph would be an option, but it doesn't meet the expectation of the isomorphic musicians i've been in contact with. they expect the duplicates to be mapped across different channels: i will try to upload the suggested mapping one of them provided to illustrate (i'm not proficient with github, i may fail to do so).

as to your last question: the number of repeats would depend on the extent of the key array. the biggest arrays i've built for testing would require no more than 5 MIDI channels.

[image]

ralfkaa, 2020-11-14:
as to why have repeat notes, and why not have a "rule" to silence repetitions: if you use two hands on a single keyboards, you may repeat a note, and you might want that to just go without issues.
i've played a bit and have found that different DAWs respond differently to a repeated not on the same MIDI channel: GarageBand will introduce a sustain, and the note will remain weird until you unplug and reconnect the keyboard. LMMS does something different, i cannot recall right now exactly what.

brandenbyers, 2020-11-14:
Ok, I think I now understand the scenario where silencing repeats wouldn’t work. Let me know if I’m understanding this correctly:

If a user presses a note on the right hand, and then presses the same note on the left hand, if the user lets up on the right hand, the left note should still be playing, right?

If the left note (the repeat of the right note) had been silenced, when lifting up on the right note, the left note would not be playing as expected.

If this scenario is correct, then I understand the need for a different solution.

brandenbyers, 2020-11-14:
Instead of channels and firmware, why not solve this problem in hardware? Wire all switches together that are the same midi key code. That way, the firmware will see each of those keys being pressed as the same key in code. So even if one of them lets up, the other(s) would still make the code think the same switch is still pressed.

Would that work? Or am I still missing something about why isomorphic users expect multiple channels?

3araht, 2020-11-14:
> Instead of channels and firmware, why not solve this problem in hardware? Wire all switches together that are the same midi key code. That way, the firmware will see each of those keys being pressed as the same key in code. So even if one of them lets up, the other(s) would still make the code think the same switch is still pressed.
> 
> Would that work? Or am I still missing something about why isomorphic users expect and/or need multiple channels?

Hi @brandenbyers, allow me to jump in.
Hardwiring is the simplest solution. However, it will kill the ability to change the keyboard layouts.
[This](https://github.com/3araht/giabalanai/blob/master/readme.md) is a keyboard I made. It is a MIDI keyboard of chromatic button accordion with no bellows. You can find an English description at the bottom.
After I released this keyboard kit, there were several layout requests to add to the new keymap layers.
[This layout list](https://github.com/3araht/giabalanai/blob/master/pictures/giabalanai_notes_layout_3araht.png) is what I have at the moment.
One of the beauties of using QMK Firmware is flexibility so hardwiring is the last choice in my opinion.

3araht, 2020-11-14:
> i've played a bit and have found that different DAWs respond differently to a repeated not on the same MIDI channel: GarageBand will introduce a sustain, and the note will remain weird until you [...]

Hi @ralfkaa, do you still have the sustain effect? It should be solved in the latest QMK firmware.
See [here](https://github.com/qmk/qmk_firmware/pull/10361) for the details of the temporary fix.

3araht, 2020-11-14:
There are many amazing isomorphic layouts!!! Here are some.

[Isomorphic keyboard](https://en.wikipedia.org/wiki/Isomorphic_keyboard)

[Chromatic Button Accordion](https://en.wikipedia.org/wiki/Chromatic_button_accordion)
- [C-System](https://en.wikipedia.org/wiki/Chromatic_button_accordion#/media/File:C-Griff.svg)
- [B-System](https://en.wikipedia.org/wiki/Chromatic_button_accordion#/media/File:B-Griff.svg)

[Chromatone](https://muto-method.com/en/chromatic-keyboard.html)

[Wicki–Hayden note layout](https://en.wikipedia.org/wiki/Wicki–Hayden_note_layout)
Check [here](https://twitter.com/s0lll0s/status/1312728372951363585?s=21) for s-ol's work, too.

[Balanced keyboard](http://www.balanced-keyboard.com)

[Jankó keyboard](https://en.wikipedia.org/wiki/Jankó_keyboard)

The strategy which throws 2nd of further duplicate notes to other channels might work with some conditions.
But might not with others.
Ideally, just simply keep the notes ON if it is pressed, would be the versatile solution.

ralfkaa, 2020-11-15:
> Ok, I think I now understand the scenario where silencing repeats wouldn’t work. Let me know if I’m understanding this correctly:
>
> If a user presses a note on the right hand, and then presses the [...]

hi @brandenbyers , this is correct for 2 of the 3 layouts i've been using: the Wicki-Hayden (first image above) and the Harmonic Table (see below). that's why these two layouts should permit activation of the same note on different keys.

but as i mentioned in my second message, this is not the case for the Jankó: for the Jankó you would want a repeat note (say, Y and B in "Mann's Virtual Jankó") to be simply ignored.

this is a complication i had honestly not thought about, nor run into yet, because the Jankó is a more linear (ascending left-to-right, like a regular piano keyboard) layout, and i am less likely to hit repeat notes by accident. it makes me think that perhaps the Jankó would have to be addressed on a different keyboard altogether (see below my answer to your other comments).

[image]

ralfkaa, 2020-11-15:
> Instead of channels and firmware, why not solve this problem in hardware? Wire all switches together that are the same midi key code. That way, the firmware will see each of those keys being pressed [...]

@3araht is aware of the issue: the keyboard in question has multiple layers. i @3araht 's case it's the accordion layouts (and, i assume, an alphanumeric layout as well). 

in my case it's at least 3 layers: alphanumeric, Wicki-Hayden, and Harmonic Table (i'm seriously considering dropping the Jankó for now, because it adds another layer of complexity, and the two MIDI layouts are sufficient for my purpose: using the keyboard as a tool to teach theory and harmony).

Jankó would be nice to have, but since the nature of the keyboard is completely different (in that you would rather have a hardware configuration such as you are describing), i will leave it out.

ralfkaa, 2020-11-15:
so i'm looking at three possibilities (2 suggested by Dean Camera, two i've though of, but don't know whether they are feasible):

1. a modifier that allows me to use "play the note X on channel Y" on a given layout, such as "channel3(note 62)" (per Dean Camera's first suggestion); i've tried variations such as (MI_CH1 | MI_C4) in layouts, but it doesn't work

2. a table that maps the layout in coordinates, to which i can assign MIDI channel AND midi note (also from Dean Camera's comment);

3. a way to assign different MIDI channels to different layers, using transparency so that a "top" layer includes notes in, say, channel 1, channel 2, and channel 3 by means of combining notes on three different layers (this is one way i imagine this could be done, but i still wouldn't know how to assign different midi channels to each layer...);

4. defining the midi notes in quantum keycodes in channel-specific terms (say, there's a C4CH1 and a C4CH2); still, i don't know if/how that can be done, because there's a midi channel set to 0 elsewhere in a "process midi" file.

there may be other/better solutions. but something along these lines would make the MIDI functionality in QMK much more robust (for Wicki-Hayden, Harmonic Table, and the C-system and B-system chromatic accordions, in particular).

ralfkaa, 2020-11-15:
> Hi @ralfkaa, do you still have the sustain effect? It should be solved in the latest QMK firmware.
> See [here](https://github.com/qmk/qmk_firmware/pull/10361) for the details of the temporary fix.

hi @3araht i haven't flashed new layouts recently, and my old mac got converted to a Fedora machine, so i'd have to give that a try. thanks for sharing.

but i still have to see what the effect is in, say, an open source DAW such as LMMS, or in online DAWs i've tested over time.

ralfkaa, 2020-11-15:
> There are many amazing isomorphic layouts!!! Here are some.
>
> [Isomorphic keyboard](https://en.wikipedia.org/wiki/Isomorphic_keyboard)
>
> [Chromatic Button [...]

i hope this isn't creating more confusion, @3araht . 

you are correct in that, as i stated previously, the Jankó needs a different solution. but the PRINCIPLE of MIDI is the ability to assign notes to up to 16 channels, so that should, IMHO, be the go-to solution for the issue of replicated notes across note arrays in isomorphic layouts.

more importantly, i researched this with professional musicians and music instructors who use isomorphic layouts extensively, and this is the solution they expect: throw duplicates onto different channels, and all is good.

ralfkaa, 2020-12-04:
hi @brandenbyers , just checking in: did you have any insights on how QMK might address this need?

ralfkaa, 2021-03-05:
closing this though this is unresolved, and a clear requirement for any more serious use of MIDI on QMK mechanical keyboards...

closed 2021-03-05 by ralfkaa