qmk/qmk_firmware #9463: BM40HSRGB Support[Feature Request]

Fenlair, 2020-06-18:
I would like to add support for the KPRepublic Hotswap BM40RGB PCB.

## Description

The BM40RGB has kbfirmware support. I'd like to port it to qmk firmware. My plan was to use https://noroadsleft.github.io/kbf_qmk_converter/, if needed fix it up so it passes the coding guidlines and add a default keymap.

Would that be acceptable?

sigprof, 2020-06-19:
This is a perfectly fine way to add support for a PCB to QMK — I did exactly this when [adding the ID80 support](https://github.com/qmk/qmk_firmware/pull/8728) (although that PR is probably more complicated than usual, because it also had some changes to optimize the somewhat unusual matrix layout, which hopefully won't be needed for BM40RGB).

One part which could bring some difficulties is that, unlike my ID80 example, BM40RGB has per-switch RGB backlight, for which the [RGB Matrix](https://docs.qmk.fm/#/feature_rgb_matrix) feature should be used (more common underglow LEDs are usually handled using [RGB Lighting](https://docs.qmk.fm/#/feature_rgblight) instead). Configuring the rgbmatrix code could be complicated; you can look at [this example of adding the BM60RGB support](https://www.reddit.com/r/MechanicalKeyboards/comments/elmt9k/porting_with_ports_kp_republic_bm60_rgb/).

Fenlair, 2020-06-21:
Thanks for the info. I set everything up and it compiles, but especially for the RGB Matrix I need the PCB first to test it.
I will get a bm40rgb in a month or two and I plan to do a pull request when I tested my patch.

sigprof, 2020-06-25:
I just realized that the Reddit link about BM60RGB that I provided does not point to the actual video (it is just a link to the Twitch channel, and that particular video is not even available in archive). See [Porting with Ports: BM60 RGB on Youtube](https://www.youtube.com/watch?v=KfPNSDZFxrY) instead.

Fenlair, 2020-06-25:
no worries I found it without an issue after I knew it existed :)

Btw if someone owns a bm40rgb and doesn't want to wait for 2 months, send me a message.

tmaj87, 2020-07-02:
> Btw if someone owns a bm40rgb and doesn't want to wait for 2 months, send me a message.
Hey, do you have working *.hex file?

PR #9860 (2020-07-30): Add KPRepublic BM40HSRGB keyboard

## Description

The current sale page for the BM40 references building firmware from kbfirmware.com and is really confusing so I just figured I'd take care of the port to the master line. I don't own one of these but helped someone set up their board and it appears to work as intended with this code set. 

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #9463

## [... cut]
PR #10044 (2020-08-18): [keyboard] update bm40hsrgb with RGB Matrix support
Updates the firmware for KPRepublic's BM40RGB keyboard that #9860 added.
This PR includes support for RGB Matrix.

## Description

This PR updates the bm40hsrgb keyboard, adding RGB Matrix support.

I've tested this bm40rgb's RGB matrix functionality with my BM40RGB keyboard.

## Types of Changes

- [ ] Core
- [ ] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [X] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [ ] Documentation

## Issues Fixed or Closed by This PR

* Fixes #9463 with RGB Matrix  [... cut]
closed 2020-07-30 by noroadsleft