qmk/qmk_firmware #9464: [Bug] Mixi didn't have default keymap, but the `default/keymap.c` exists

ohchiko, 2020-06-18:
QMK Configurator said that Mixi keyboard didn't have default keymap yet. But the folder `mixi/keymaps/default/keymap.c` exists.

## Describe the Bug
Opens [QMK Configurator](https://config.qmk.fm/), select _mixi_ on the _Keyboard_ options, all keys showing `N/A` and only layer 0 is activated. Click on _Load Default_, the log said `* Sorry there is no default for the mixi keyboard... yet!`. Then I checked the commit files, and found the `keymap.c` file on the default keymap folder. Below is the screenshot:
[image]

## System Information

 - Keyboard:  Mixi
   - Revision (if applicable):  
 - Operating system: Windows
 - AVR GCC version: 8.3.0

 - ARM GCC version: 8.3.1 20190703 (release)

 - QMK Firmware version: 0.9.12

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

ohchiko, 2020-06-22:
Problem still exists after new [PR](https://github.com/qmk/qmk_firmware/pull/9467) merged.

fauxpark, 2020-06-22:
Default keymaps for the Configurator are added [here](https://github.com/qmk/qmk_configurator/tree/master/public/keymaps).

ohchiko, 2020-06-22:
Oh never thought I should PR to the Configurator repo too.
PR submitted.

ohchiko, 2020-06-23:
PR merged, problem fixed. Thank you.

closed 2020-06-23 by ohchiko