qmk/qmk_firmware #9503: [Feature Request] Counting Key

AlexOConnorHub, 2020-06-22:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [X] New behavior

## Description
As a Computer Science student, I do a fair amount of coding. When i saw the dynamic macro functions in QMK, I thought about how nice it would be for some of the repetitive copy/paste parts of coding. To help expand this, would a counting key be possible? The idea being that when pressed, it outputs `0`, next time it is pressed, it outputs `1`, and so on. Being combined with a dynamic macro could allow for switch/case statements to be quickly outlined, just to name one of the many ways this could be used. For my keyboard, I am working on making this a key, as I know I will get good usage out of such a key. Would such a key be something that might be worth adding to QMK?

tyetye, 2020-06-23:
Would the [Tap Dance](https://docs.qmk.fm/#/feature_tap_dance) feature work for your needs?

AlexOConnorHub, 2020-06-23:
This would not, because tap dance still executes the same action after the same combo /set of taps. This key would execute differently each time tapped.

rmwphd, 2020-06-24:
This sounds almost more like an IDE feature than a keyboard feature. Can you give some more examples of the kind of functionality this would enable?

AlexOConnorHub, 2020-06-29:
So, just trying to think other usages: 
* Programming case statements
* Renaming a batch of files with ascending numeral order
* Numbering in a word doc (if you cannot use default numbering)

I guess I cannot think of many uses. This just might be more of an IDE feature like you said, I was just thinking of ways to expand on dynamic macros.

closed 2020-07-22 by AlexOConnorHub