qmk/qmk_firmware #9505: [Bug] Can't get OS key to work

matteolecuit, 2020-06-22:
Can't get OS key to work.

## Describe the Bug

I used qmk configurator to make the keymap and every key works except the OS key. I tried with LEFTOS/RIGHTOS, in both ANSI and ISO, but the key doesn't do anything when pressed. I though it could be an issue with my pcb or switch, so I tried to map the key to another action (a letter), and the key works.

## System Information

 - Keyboard:  preonic
   - Revision (if applicable):  rev3
 - Operating system: Ubuntu Linux 20.04 (kernel 5.4.0)
 - AVR GCC version: 5.4.0
 - ARM GCC version: 9.2.1
 - QMK Firmware version: 0.9.19
 - Any keyboard related software installed? 
none

## Additional Context

I previously flashed the board a few month ago, and OS key worked.

AlexOConnorHub, 2020-06-22:
So the OS key works when you have it mapped to a key other than what it has been for the past few months. It sounds like it is a PCB or switch problem. try jumping the contacts where the OS switch usually is, and see if it registers as a key press.

matteolecuit, 2020-06-22:
> So the OS key works when you have it mapped to a key other than what it has been for the past few months. It sounds like it is a PCB or switch problem. try jumping the contacts where the OS switch usually is, and see if it registers as a key press.

No, i probably wasn't clear. OS key doesn't work anywhere on the board, and the key I wanted map to OS works, as I tried to put a letter in its place.

AlexOConnorHub, 2020-06-22:
Oh, my bad there. That makes more sense though.

fauxpark, 2020-06-22:
Sounds like it could be the work of [Bootmagic](https://docs.qmk.fm/#/feature_bootmagic) - try holding space+backspace while plugging it in.

matteolecuit, 2020-06-23:
> Sounds like it could be the work of [Bootmagic](https://docs.qmk.fm/#/feature_bootmagic) - try holding space+backspace while plugging it in.

This indeed fixed the issue. Thanks a lot, I had no idea Bootmagic existed, I guess I should've read the doc more thoroughly.

closed 2020-06-23 by matteolecuit