qmk/qmk_firmware #9506: Trying to get arm_atsam.mk working with SAMD21 processor

cbc02009, 2020-06-22:
I'm trying to get QMK working with a handwired macropad I'm working on. The macropad runs on a SAMD21G18A, but the makefiles seem to have been written assuming someone would only ever use the SAMD51 processor.

I'm trying to change arm_atsam.mk includes to conditionally compile using the $(MCU) define, like so:
`ifeq ($(MCU),SAMD51J18A)
COMMON_VPATH += $(LIB_PATH)/arm_atsam/packs/atmel/SAMD51_DFP/1.0.70/include
LDFLAGS += -T$(LIB_PATH)/arm_atsam/packs/atmel/SAMD51_DFP/1.0.70/gcc/gcc/samd51j18a_flash.ld
endif

ifeq ($(MCU),SAMD21G18A)
COMMON_VPATH += $(LIB_PATH)/arm_atsam/packs/atmel/SAMD21_DFP/1.3.395/include
LDFLAGS += -T$(LIB_PATH)/arm_atsam/packs/atmel/SAMD21_DFP/1.3.395/gcc/gcc/samd21g18a_flash.ld
endif`

but I have a feeling that when this code is reached, $(MCU) hasn't been defined yet, as it doesn't seem to work.

I could create a separate arm_atsamd21 protocol, but as the atmel software framework library is written to work with multiple processors, it seems like there would be a lot of code duplication added just to get around this one issue.

cbc02009, 2020-06-22:
Oops, $(MCU) is supposed to be $(ARM_ATSAM). That was dumb of me.

closed 2020-06-22 by cbc02009

fauxpark, 2020-06-23:
I'm not sure that we would accept additions to the ATSAM code, to be honest. It was mainly contributed by Massdrop for their boards, and is not very well maintained (hence there is only one chip supported). I think we would much rather see support for the ATSAM family added to ChibiOS, or integration of [RIOT-OS](https://www.riot-os.org/) which would supposedly give us much more than ATSAM support.

cbc02009, 2020-06-23:
Yeah, the more I look at it, the more I realize it's a near impossible task to extricate the massdrop keyboards from the ATSAM code. Very disappointing. I'll look into the other options and see if anything else is doable.

cbc02009, 2020-06-23:
@fauxpark Is there a porting guide for QMK? RIOT seems close enough to freeRTOS (which I know pretty well) that I might be able to get it working without having to create an entirely new HAL like I would for chibiOS.

okyeron, 2020-08-26:
@cbc02009 Any further movement here? I'm working on a similar macropad with the SAMD21G18A.

FWIW - RIOT-OS appears to support SAMD21 
https://github.com/RIOT-OS/RIOT/tree/master/boards/samd21-xpro
> The `SAMD21 Xplained Pro` is an ultra-low power evaluation board by Atmel
featuring an ATSAMD21J18A SoC. The SoC includes a SAMD21 ARM Cortex-M0+ micro-
controller. For programming the MCU comes with 32Kb of RAM and 256Kb of flash
memory.

cbc02009, 2020-08-26:
@okyeron I've started working on a port for RIOT, but I have literally no experience whatsoever with Make (we use CMake at work), so I'm not sure how far I'll be able to get. So far the makefiles for the two projects don't seem to play very well with each other.