qmk/qmk_firmware #9509: Using ILI9341 with uGFX

HalFrgrd, 2020-06-22:
Hi,

I have an ILI9341 (using SPI) that I want to use with my pro micro running QMK. I think I'd like to use uGFX but I can't figure out how. Could someone provide some guidance please?

Thank you

tzarc, 2020-06-24:
@HalFrgrd we're actually working towards the removal of uGFX over time, as the licensing of the library is incompatible with the distribution of binaries that include it.

Fear not, I've actually been working on an ILI9341 driver specifically for QMK -- just need some time to finalise it: [image]

HalFrgrd, 2020-06-24:
Thank you, looking forward to it!

closed 2022-08-04 by tzarc