qmk/qmk_firmware #9521: [Bug] Custom keycode now clears the one shot layer

plattfot, 2020-06-23:
## Describe the Bug
Have a double tap function in my [keymap](https://github.com/qmk/qmk_firmware/tree/master/keyboards/kyria/keymaps/plattfot), with the commit 9c1097e768aed90e4c56a3a18f6c1cc9cac52391 it now clears my one shot layers, where as before it didn't. The expected behavior for it is to not clear the one shot layer as I mainly use this double tap key to repeat symbols on my symbol layer. Where I'm using my right thumb to first jump to the symbol layer with a one shot layer then with the same thumb press the double tap key.

Here's the code for the double tap function 
```
//...
static bool double_tap_it = false;
//...
bool process_record_user(uint16_t keycode, keyrecord_t *record) {
    if (record->event.pressed) {
        switch (keycode) {
            //...
            case DBL_TAP:
                double_tap_it = !double_tap_it;
                return false;
           //...
        }
    } else if (double_tap_it && keycode != DBL_TAP) {
        double_tap_it = false;
        tap_code16(keycode);
        clear_oneshot_layer_state(ONESHOT_OTHER_KEY_PRESSED);
    }

    return true;
}
``` 

## System Information

 - Keyboard:
   Kyria  
   - Revision (if applicable):
    1.1
   - Operating system:
    Arch Linux - linux kernel: 5.6.15-arch1-1
 - AVR GCC version: 

10.1.0
 - ARM GCC version: 

 
- QMK Firmware version:
0.8.44

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
Ran a git bisect on a working commit and master. This commit was the one that introduced the bug for me: 9c1097e768aed90e4c56a3a18f6c1cc9cac52391

spidey3, 2020-06-23:
The change in behavior is due to this block of code:
https://github.com/qmk/qmk_firmware/blob/9c1097e768aed90e4c56a3a18f6c1cc9cac52391/tmk_core/common/action.c#L195-L202

Lines 196-200 were added in #8832 .

The question is, under what circumstances should we be cleaning up the OSL after `process_record_quantum()` (which in turn eventually calls `process_record_user()`? Should it be cleaned up only true is returned? 

It doesn't seem right to me that a simple macro implemented in `process_record_user()` should have to worry about cleaning up OSL state — but on the other hand, I'm also not sure that we should prevent @plattfot 's use case...

spidey3, 2020-06-23:
Maybe there needs to be a way to flag whether OSL should be cleaned up after `process_record_user()`, separate from the return value...

sigprof, 2020-06-24:
I tried to add a way to configure the OSL reset behavior in #9538 — what do you think about this solution?

drashna, 2020-07-05:
To be blunt, the new behavior is the expected behavior. It should be cleared after any keypress.  That it didn't .... was a bug in the cleanup. 

Personally, I don't feel that the behavior should be reverted/fixed.  In this case, tap toggle may be a better way.

plattfot, 2020-07-05:
For me it would be great to opt out, so I can still have my double tap feature. But I understand why people don't want to enable that. And if that's is the case I'll just have learn to live without it.

Tap toggle for me is worse, as that would result in me needing to press more keys than just holding down the layer key and press a key twice and I'd always trouble syncing my key presses if I wanted to quickly type a symbol then jump back to my default layer. A better way for me is either to remove the feature entirely and just embrace the little bit less comfy way of typing double symbols. Or add a macro layer with the double symbols, not as elegant or as flexible as what I have right now. And I'm not sure I have enough space left on the EPROM for it. But if it works I'll reduce the key presses for typing double symbols.

plattfot, 2020-07-06:
I realized a better way for me to use double tap that doesn't require turning off the clearing of the one shot layer, and that is to not clear double tap when I press any of the layer keys. So now I can first press double tap, then jump to a one shot layer and press the key for a specific symbol.

With pull request #9666, this is no longer an issue for me.

egst, 2020-12-01:
I think this could be the reason I couldn't implement oneshot functionality manually. It doesn't matter whether I return true or false from `process_record_user`, it always clears the `OTHER_KEY_PRESSED` state and as soon as I lift the key and clear the `ONESHOT_KEY_PRESSED` state manually, my custom oneshot layer is turned off. This basically prevents any attempt to implement oneshot functionality using `set_oneshot_layer` and `clear_oneshot_layer_state`, so if there is no other way the oneshot functionality is supposed to be implemented, I think this should be addressed.

mcp292, 2021-06-12:
I'm trying to create a workaround for this:
```c
#define SFT_TOP LM(TOP, MOD_LSFT)
```

Where `TOP` is my OSL that I want to use the shift key in. While this key works outside the `TOP` layer, when pressed inside it seems to return you to `BASE`.