qmk/qmk_firmware #9627: [Bug] Keymaps for a keyboard can't be compiled after "2020 May 30 Breaking Changes Update".

blockader, 2020-07-02:
After "2020 May 30 Breaking Changes Update", the keymaps for keyboards/matrix/noah can't be compiled. As a user of this keyboard, I am not able to use the master branch now.

## Describe the Bug

git checkout master
qmk compile -kb matrix/noah -km default
This should give the following error:
keyboards/matrix/noah/matrix.c:12:10: fatal error: printf.h: No such file or directory
I tried to fix this but I am not familiar with the code so I am asking for help.

## System Information

 - Keyboard:  Matrix Noah
   - Revision (if applicable):  
 - Operating system: macOS Catalina
 - AVR GCC version: 8.4.0

 - ARM GCC version: 8.3.1

 - QMK Firmware version: 0.8.148

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2020-07-02:
That error is probably because of this: https://docs.qmk.fm/#/ChangeLog/20200530?id=upgrade-from-tinyprintf-to-mpalandprintf

blockader, 2020-07-02:
> That error is probably because of this: https://docs.qmk.fm/#/ChangeLog/20200530?id=upgrade-from-tinyprintf-to-mpalandprintf

Thanks. I followed this document and applied "make git-submodule". It seems that the problem is now fixed.

closed 2020-07-02 by blockader